RetroPie/RetroPie-Setup #905: Retroarch crash when changing shaders in game

biscuits99, 2015-07-12:
This works in beta 2, fails in beta 4.

When in game, holding the enable hotkey (select) then pressing m or n to scroll shader causes Retroarch to crash and return to Emulation Station.

I thought the problem may be related to paths or permissions, but I cant seem to prove that with an example. The issue does not happen with other hotkeys on keyboard such as 'h' or 'F1'. So I'm guessing it is someway file/path/permission related.

Details here:
http://blog.petrockblock.com/forums/topic/hotkeys-configuration-switch-the-shaders-ingame/

joolswills, 2015-07-13:
did you test updating to the latest retroarch binary via menu 5 ?

joolswills, 2015-07-13:
Also if you build retroarch from source, does that help ?

joolswills, 2015-07-14:
Unable to reproduce with the current retroarch - if you are testing with the version of retroarch from the last image, then it has multiple known issues, so it is recommended to update.

biscuits99, 2015-07-14:
From beta 4 image, I updated retroarch from binary. Then no problem at all.
Not sure if updating script would also be recommended, but above step alone fixed it for me.

joolswills, 2015-07-15:
ok cool. Ill close this then. retroarch just tagged 1.2.1 so I might lock down the retroarch to that version due to the number of problems we had recently during the development.

closed 2015-07-15 by joolswills