RetroPie/RetroPie-Setup #906: imame4all-libretro slows itself down by outputting too many messages to the console - fix within!

dankcushions, 2015-07-16:
eg Robotron 2084 (note: you have to select 'Content Settings' > 'Restart Content' in the retroarch GUI after you get the "Factory Settings Restored" message, else it will hang) - you'll notice that when the player spawns at the start of the game, it slows to a crawl. The reason for this is it's logging a load of error messages to the console. If you instead amend /opt/retropie/configs/mame-mame4all/emulators.cfg to
 lr-imame4all="/opt/retropie/emulators/retroarch/bin/retroarch > /dev/null -L /opt/retr(etc...)
...it doesn't display these messages, and the game plays flawlessly. Apparently this affects other games (outrun, others?)

I realise this is a problem with the core (or maybe retroarch itself?), so I've logged an issue there (https://github.com/libretro/imame4all-libretro/issues/12), but the project hasn't been updated in a year, so it might be better to do this fix (or some other way to stop it overloading the system?)

joolswills, 2015-07-16:
fixed - I forked the code and disabled logging in the code. Building updated binaries now.

closed 2015-07-16 by joolswills

HerbFargus, 2015-07-17:
I can confirm this is resolved. Thanks.

dankcushions, 2015-07-17:
Thank you! How do I get the latest version? apt-get and then rebuild from binaries?

joolswills, 2015-07-17:
just reinstall from binary in menu 5

dankcushions, 2015-07-17:
Tested and works - great job :)