RetroPie/RetroPie-Setup #908: pifba crash

joolswills, 2015-07-17:
I updated the setup script and installed from binary- it acts like its loading but then it exits right back out. runcommand gives this error- this game was previously working with pifba before the update (cadillacs and dinosaurs)

For the parent
```
/opt/retropie/supplementary/runcommand/runcommand.sh: line 684:  3130 Segmentation fault      /opt/retropie/emulators/pifba/fba2x "/home/pi/RetroPie/roms/fba/dino.zip"
```
and the clone
```
/opt/retropie/supplementary/runcommand/runcommand.sh: line 684:  2847 Segmentation fault      /opt/retropie/emulators/pifba/fba2x "/home/pi/RetroPie/roms/fba/dinoj.zip"
```

The game runs fine in lr-fba

joolswills, 2015-07-17:
@HerbFargus just that game btw or ?

joolswills, 2015-07-17:
Unable to reproduce -that rom  working here on pifba fine (from mame romset 0.37b5)

HerbFargus, 2015-07-17:
Its happening to every game that previously worked. I tried rebuilding from source to see if that made any difference. It kept throwing this warning 
```
missing braces around initializer
```

joolswills, 2015-07-17:
you can ignore that.

Maybe it's something in the config ? I have updated the binaries, and they work fine on my system here - tested with a bunch of roms included the one you mention. you can try with a new image shortly perhaps (Check g+)

HerbFargus, 2015-07-17:
That's what I was thinking, I'll give the release candidate a go once you've uploaded the image and that should eliminate my potential user error.

joolswills, 2015-07-17:
should be ready in 30 mins or so.

HerbFargus, 2015-07-17:
Ok, so I've tested it and I think it may have something to do with my configs- I logged in as root on winscp and changed the fba2x.cfg so that my select and start would work, and as soon as I saved it, it didnt work anymore. I don't know if its because it changed the owner to root instead of pi- I chowned it back to user and group pi but now it just doesn't work anymore. I'll try modifying it from the terminal on a fresh build and see if it has the same result.

HerbFargus, 2015-07-17:
Nope- changing from the terminal is no different. anytime I make any changes to fba2x.cfg it no longer works. perhaps its related to this commit?:

https://github.com/RetroPie/RetroPie-Setup/commit/34cb86efaf9cfd49b16c2ab7c4f3248bb99fbe65

joolswills, 2015-07-17:
Please can you detail what changes you made so I can reproduce ?

I doubt it's connected to that commit, as it works fine with the default config - it's likely to do with upstream fba2x changes perhaps - but if you give me your config I can test here.

joolswills, 2015-07-17:
permissions are wrong on the default configs though so I'll fix that.

HerbFargus, 2015-07-17:
All I changed was the numbers on start and select from 6 and 7 to 8 and 9 on all 4 players

joolswills, 2015-07-17:
It would be easier for me to see a copy of the config please. Also by default it has 2 configs for keyboard and 2 for joystick so I'm not exactly sure how to reproduce your config.

HerbFargus, 2015-07-17:
this is my /opt/retropie/configs/fba/fba2x.cfg
```
[Keyboard]
# Get codes from /usr/include/SDL/SDL_keysym.h
A_1=306
B_1=32
X_1=308
Y_1=304
L_1=122
R_1=120
START_1=13
SELECT_1=9
LEFT_1=276
RIGHT_1=275
UP_1=273
DOWN_1=274
QUIT=27
#player 2 keyboard controls, disabled by default
A_2=999
B_2=999
X_2=999
Y_2=999
L_2=999
R_2=999
START_2=999
SELECT_2=999
LEFT_2=999
RIGHT_2=999
UP_2=999
DOWN_2=999

[Joystick]
# Get codes from "jstest /dev/input/js0"
# from package "joystick"
# Defaults to XBOX 360 Controller
A_1=1
B_1=0
X_1=2
Y_1=3
L_1=4
R_1=5
START_1=9
SELECT_1=8
#Joystick axis
JA_LR=0
JA_UD=1
#player 2 button configuration
A_2=1
B_2=0
X_2=2
Y_2=3
L_2=4
R_2=5
START_2=9
SELECT_2=8
#Joystick axis
JA_LR_2=0
JA_UD_2=1
#player 3 button configuration
A_3=1
B_3=0
X_3=2
Y_3=3
L_3=4
R_3=5
START_3=9
SELECT_3=8
#Joystick axis
JA_LR_3=0
JA_UD_3=1
#player 4 button configuration
A_4=1
B_4=0
X_4=2
Y_4=3
L_4=4
R_4=5
START_4=9
SELECT_4=8
#Joystick axis
JA_LR_4=0
JA_UD_4=1

[Graphics]
DisplaySmoothStretch=1
# Display Effect: 0 none, 1 scanlines
DisplayEffect=0
DisplayBorder=0
MaintainAspectRatio=1
DisplayAutoRotate=0

[Sound]
```

Default was 
```
[Keyboard]
# Get codes from /usr/include/SDL/SDL_keysym.h
A_1=306
B_1=32
X_1=308
Y_1=304
L_1=122
R_1=120
START_1=13
SELECT_1=9
LEFT_1=276
RIGHT_1=275
UP_1=273
DOWN_1=274
QUIT=27
#player 2 keyboard controls, disabled by default
A_2=999
B_2=999
X_2=999
Y_2=999
L_2=999
R_2=999
START_2=999
SELECT_2=999
LEFT_2=999
RIGHT_2=999
UP_2=999
DOWN_2=999

[Joystick]
# Get codes from "jstest /dev/input/js0"
# from package "joystick"
# Defaults to XBOX 360 Controller
A_1=1
B_1=0
X_1=2
Y_1=3
L_1=4
R_1=5
START_1=6
SELECT_1=7
#Joystick axis
JA_LR=0
JA_UD=1
#player 2 button configuration
A_2=1
B_2=0
X_2=2
Y_2=3
L_2=4
R_2=5
START_2=6
SELECT_2=7
#Joystick axis
JA_LR_2=0
JA_UD_2=1
#player 3 button configuration
A_3=1
B_3=0
X_3=2
Y_3=3
L_3=4
R_3=5
START_3=6
SELECT_3=7
#Joystick axis
JA_LR_3=0
JA_UD_3=1
#player 4 button configuration
A_4=1
B_4=0
X_4=2
Y_4=3
L_4=4
R_4=5
START_4=6
SELECT_4=7
#Joystick axis
JA_LR_4=0
JA_UD_4=1

[Graphics]
DisplaySmoothStretch=1
# Display Effect: 0 none, 1 scanlines
DisplayEffect=0
DisplayBorder=0
MaintainAspectRatio=1
DisplayAutoRotate=0

[Sound]
```

joolswills, 2015-07-17:
i may have had an old config lying around. Ill check and compare thanks.

joolswills, 2015-07-17:
I copied your config over and it worked fine. You didn't introduce some mixed line endings or something perhaps ?

joolswills, 2015-07-17:
```
$ ls -l /opt/retropie/configs/fba
total 16
-rw-r--r-- 1 pi pi  195 Jul 17 16:25 capex.cfg
-rw-r--r-- 1 pi pi   66 Jul 17 16:25 emulators.cfg
-rw-r--r-- 1 pi pi 1070 Jul 17 16:26 fba2x.cfg
-rw-r--r-- 1 pi pi 1069 Jul 17 16:26 fba2x.cfg.original
```

joolswills, 2015-07-17:
```
pi@retropie /opt/retropie/configs/fba $ diff -ru fba2x.cfg.original fba2x.cfg
--- fba2x.cfg.original	2015-07-17 16:26:53.797140292 +0000
+++ fba2x.cfg	2015-07-17 16:31:30.331937799 +0000
@@ -37,8 +37,8 @@
 Y_1=3
 L_1=4
 R_1=5
-START_1=7
-SELECT_1=6
+START_1=9
+SELECT_1=8
 #Joystick axis
 JA_LR=0
 JA_UD=1
@@ -49,8 +49,8 @@
 Y_2=3
 L_2=4
 R_2=5
-START_2=7
-SELECT_2=6
+START_2=9
+SELECT_2=8
 #Joystick axis
 JA_LR_2=0
 JA_UD_2=1
@@ -61,8 +61,8 @@
 Y_3=3
 L_3=4
 R_3=5
-START_3=7
-SELECT_3=6
+START_3=9
+SELECT_3=8
 #Joystick axis
 JA_LR_3=0
 JA_UD_3=1
@@ -73,8 +73,8 @@
 Y_4=3
 L_4=4
 R_4=5
-START_4=7
-SELECT_4=6
+START_4=9
+SELECT_4=8
 #Joystick axis
 JA_LR_4=0
 JA_UD_4=1
```

joolswills, 2015-07-17:
Just in case, did you try without a controller connected in case it's something to do with that too ?

HerbFargus, 2015-07-17:
Found the issue. Had 3 controllers plugged in- it works fine with just one controller.... hmm. Sorry I should have mentioned my hardware setup at the beginning. the 3 controllers are exactly the same.

joolswills, 2015-07-17:
Guess it needs some more work for the 4 player support. I think this will have to be done after the release candidate though.

HerbFargus, 2015-07-17:
Sounds good- yep it works fine with two, crashes with 3.  I should have tested that sooner- my bad- thanks.

joolswills, 2015-07-17:
no worries. we can just make a note of that on the wiki or something. I'll compare the upstream code with the patch from digitallumberjack, and go from there.

veteranjohnston, 2015-08-01:
HerbFargus, thank you so so so so so much, I have been pulling my hair out trying to figure out WTF happened, I had the same problem as you and never even thought to unplug a controller. It is strange though cause before I changed around some of my controls It worked fine with 3 controllers, plugged it

joolswills, 2016-03-15:
should be fixed now (works here) - I'm rebuilding binaries.

HerbFargus, 2016-03-15:
:D really? Will test as soon as they are built!

joolswills, 2016-03-15:
built/uploaded.

HerbFargus, 2016-03-15:
TESTED AND FINALLY FIXED! HOORAY

closed 2016-03-15 by HerbFargus