RetroPie/RetroPie-Setup #909: Crash in Videopac emulator in 3.0 RC1 and in source build from 7/18/2015 with Pi2

truckershitch, 2015-07-19:
I was getting the same error in 3.0 RC1 so I build the project from source from the Retropie-Setup script last night.  This is on a Pi2.  I did not have a problem with 3.0 beta4.

/opt/retropie/supplementary/runcommand/runcommand.sh: line 684:  3873 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-o2em/o2em_libretro.so --config /opt/retropie/configs/videopac/retroarch.cfg "/home/pi/RetroPie/roms/videopac/Attack_of_the_Timelord_(USA).bin" --appendconfig /tmp/retroarch.cfg

I have run backtraces and the like before but I will need some help to produce them.

Thanks!

truckershitch, 2015-07-19:
Ok, false alarm.  I didn't have o2rom.bin in the bios directory.  Still, I don't think it should have segfaulted.

Sorry for the noise.

closed 2015-07-19 by truckershitch

joolswills, 2015-07-19:
You can report the problem with it segfaulting to https://github.com/libretro/libretro-o2em