RetroPie/RetroPie-Setup #911: RetroPie 3.0 RC1

lip5016, 2015-07-22:
I think it's kind of crappy that you still can't get a PS3 controller to work wirelessly without having to run command lines to set them up first. You should be able to plug in your PS3 controller 1 time when you turn it on, configure it at the default EmulationStation screen that comes up, and then be able to take your wire out and have it continue to work wirelessly (in a similar fashion that a regular PS3 would do). 

Countless people out there have PS3 controllers and would love for this whole process to be a lot easier and more streamlined. Please consider this! Thank you!!

joolswills, 2015-07-22:
I don't have a ps3 controller - not sure if any other core devs have one either - happy for someone to contribute this functionality.

Despite you thinking it is crappy - we do this project for free in our spare time. If this is your best go at a feature request, you might want to consider rewording it.

lip5016, 2015-07-22:
I'm sorry, you're right. 

I've just had really high expectations for RetroPie. Most of them have been met, exceedingly. The one thing that always seems to give me issues is pairing multiple wireless PS3 controllers. I just wish it was as painless as pairing with the actual console. I respect what you guys do, if you think at some point one of you are able to get your hands on a PS3 remote and work this out, I'm sure there will be a lot more people besides me extremely happy for the contribution.

cheers :)

gizmo98, 2015-07-22:
Do you mean ps3controller module should be installed by default? Back in 3.0 beta 2 days plug in and pair was possible with this module. Don't know if it is still working. I have heard of some issues.

joolswills, 2015-07-22:
It should still be working - or at least nothing has been broken intentionally :)

gizmo98, 2015-07-22:
Floob had written me a message suggesting something like this. Sorry Floob, i have not answered till know :D. It should work if raspbians bluetooth daemon is still the same.

lip5016, 2015-07-22:
The whole thing is that I'm able to map and assign controls to my PS3 controlled if I have the wire connected, and leave it connected, but once I remove the wire, the lights on the PS3 controller just keep blinking and don't ever actually connect wirelessly. I know a lot of my friends would love to have their own RetroPie but most of them don't have the computer skills to sit there and mess around with command line/SSH/etc. I really would love a quick and easy 'finished product' that I can set my friends up with, so that they can plug it in and be off to a running start. Hopefully eventually things will be this way with RetroPie! :)

ProxyCell, 2015-08-07:
hey i have some authentic PS3 controllers, what else would i need to get this working? maybe i can figure out a solution for you lip - i just need you to help and work with me

joolswills, 2015-08-13:
I have a ps3 controller being sent to me, so I will hopefully improve the ps3 controller pairing etc.

ProxyCell, 2015-08-13:
@joolswills yay! i have no intention of ever using one with the RetroPie really but hopefully this will reduce the terrifying amount of tickets/threads/patches/etc related to it

lip5016, 2015-08-13:
@ProxyCell you da real MVP! :100: 

you have no idea how much this is appreciated!

some thoughts about what would be helpful:

-easy connect via bluetooth.. just like when you first start up a PS3, you plug in the controllers once to pair, and then as soon as you press the "PS" button.. (aka. the circle button in the center of the PS3 controller) it pairs to the LED slot on the top of the controller (for players 1-4), and then you can easily unplug the controller and continue to operate completely wirelessly via bluetooth

-ability to connect 2+ controllers wirelessly

-ability to "exit" emulators by clicking the "PS" button.. i think it's currently set as pressing "Select + PS", but I think just pressing the "PS" button alone should be more than sufficient

once again, I really appreciate your help and thank you for hearing out my suggestions! i can bet you that this will help out countless users and make RetroPie the super-sweet-super-dope console for even the most novice user! :dancers:

ProxyCell, 2015-08-13:
hey @lip5016 - please thank @joolswills  ;)

joolswills, 2015-08-13:
There was a minor logic error in the script for paring, but @lip5016 did you try the retropie-setup menu option to install/pair ps3. I've tested it and it does work - connecting by usb, then unplugging, and pressing the playstation button and it connects.

joolswills, 2015-08-13:
regarding exiting without a hotkey, this is only possible in the case that you dont want any other hotkeys to work while pressing select. currently, you need to either have a hotkey button or not, and exiting is a hotkey.

closed 2015-10-08 by joolswills