RetroPie/RetroPie-Setup #912: Have to unplug F310 controller when I run emulator (latest release).

olanin, 2015-07-23:
I am using RetroPie SD-card Image for Raspberry Pi 2 Version 3.0 RC 1, and having a problem with my Logitech F310 controller. This is the only controller I use with retropi. I went through the automatic setup after I imaged the card. Each time I turn retropi on, the controller works fine with the main menu (choosing which emulator, which rom, ect..), but when I launch a rom, the controller does not work. I have to unplug it (and on the screen it says "Device 0 unplugged Logitch F310), and when I plug it back in (Logitech F310 configured in port) and it works. It does this each time and I even re imaged and set it up again.

joolswills, 2015-07-29:
I don't have this controller, so will be unable to advise. Not seen this behaviour with any of my controllers either. Perhaps you can find another user with this controller on the forum and see if they are having the same problem ?

olanin, 2015-07-29:
I wanted to use the F310 the most. Its all I have right now anyways. Since I posted this, I went back to version 2.6 and it works perfectly everytime. It must be that something was changed in the latest release. I also notice that the savestate keycombo does not work with 3RC1, and when I hit select it says load state failed for some reason.

joolswills, 2015-07-29:
it must be related to Retroarch changes I guess - there is nothing in our project that would have changed this behaviour. Please go to menu 5 and tro rebuilding retroarch from source and see if that changes anything (to get the very latest version).

Also try launching retroarch from console without emulationstation

```
/opt/retropie/emulators//retroarch/bin/retroarch -L /opt/retropie/libretrocores/path/to/libretro.so --config /opt/retropie/configs/all/retroarch.cfg <path to rom>
```

and see it happens when just running retroarch.

joolswills, 2015-07-31:
Have you tried the above ?

sandersch, 2015-08-01:
I have an F310 and just tested with a fresh image on a Raspberry Pi 2. The controller worked correctly for me (although the setup in EmulationStation is a little touchy with the bottom triggers) in both ES and in multiple emulators.

@olanin do you have your controller plugged directly into the Pi or are you using a powered USB hub? I used to have issues with this controller being flakey with my Pi until I started using a powered hub. I think this controller is pretty sensitive to power issues. Might be worth giving a try.

joolswills, 2015-08-01:
Good point - I probably should have mentioned that. A lot of USB issues are down to PSU - a recent person had their pi resetting when they pressed a key, and it ended up being they were powering the pi via a usb cable from a computer usb port (Which may well have been limited to 500mA)

joolswills, 2015-08-10:
Closing due to no feedback from OP.

closed 2015-08-10 by joolswills

EBCool8, 2015-08-25:
Please reopen this issue. I have both a Logitech F310 and F510 and both have issues with the trigger buttons not being able to be defined properly in the settings menu. I'm enjoying the 3.0 release for my Pi 2 (although a vast number of my SNES roms are not playable) but it's killing me not being able to define triggers as well as having problems with save and load state hot keys not working even though increment/ decrement  state slot, exit, and rom menu shortcuts all work. This is a serious hampering for those of us who have used Logitech controllers for years and want to continue to enjoy the portability of a Pi 1/2 with Retropie without adding a powered USB hub to the equipment carried or having to change controllers; although I am about try my PS3 controllers to see if I can finally save/load state, I do worry about the required power for a PS3 controller versus a F510 after reading the comments about power issues.

joolswills, 2015-08-25:
please open a new ticket.

olanin, 2015-08-25:
My fix has been to use the previous version. Works great. I dont have time now to play around with the current version, but will try the next release whenever it comes out.