RetroPie/RetroPie-Setup #913: Add Dragon / Tandy Color Computer Emulation

Bosco70, 2015-07-23:
Would it be possible to add Dragon / Tandy Color Computer emulation?

XRoar supports both and has been ported to the RPi.

http://archive.worldofdragon.org/phpBB3/viewtopic.php?f=5&t=754

Thanks.

joolswills, 2015-07-27:
I've almost got this working. Just a few final details to sort.

joolswills, 2015-07-27:
added - you can find xroar in the experimental section.

HerbFargus, 2015-07-28:
I tried to build it from the experimental menu and I get the error- could not successfully build_xroar Dragon / Coco Emulator XRoar (/home/pi/RetroPie-Setup/tmp/build/xroar/xroar not found). but xroar is there in the build folder. This is the full error log

```
Log started at: Tue Jul 28 23:27:59 UTC 2015

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'xroar' : Dragon / CoCo emulator XRoar
= = = = = = = = = = = = = = = = = = = = =

/home/pi/RetroPie-Setup/tmp/build/xroar /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'xroar' : Dragon / CoCo emulator XRoar
= = = = = = = = = = = = = = = = = = = = =

git clone --branch rasppi "http://www.6809.org.uk/git/xroar.git" "/home/pi/RetroPie-Setup/tmp/build/xroar"
Cloning into '/home/pi/RetroPie-Setup/tmp/build/xroar'...
/home/pi/RetroPie-Setup
/home/pi/RetroPie-Setup/tmp/build/xroar /home/pi/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Building 'xroar' : Dragon / CoCo emulator XRoar
= = = = = = = = = = = = = = = = = = = = =

for i in ilclient.o ilcore.o; do (if test -e "$i"; then ( rm $i ); fi ); done
cc -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I../libs/ilclient -I../libs/vgfont -g -c ilclient.c -o ilclient.o -Wno-deprecated-declarations
cc -O2 -mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -pipe -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I../libs/ilclient -I../libs/vgfont -g -c ilcore.c -o ilcore.o -Wno-deprecated-declarations
ar r libilclient.a ilclient.o ilcore.o
ar: creating libilclient.a
rm ilcore.o ilclient.o
Build system type is arm-linux-gnueabihf
Target system type is arm-linux-gnueabihf
Checking for C99 ...  -std=c99
Checking for Objective C optimisation options ... 
Checking for Endianness ... little
Checking for gcc -std=c99 accepting -pipe ... yes
Checking for g++ accepting -pipe ... yes
Checking for gcc accepting -pipe ... no
Checking for strings.h ... yes
Checking for strcasecmp() ... yes
Checking for strsep() ... yes
Checking for zlib ... yes
Checking for GLib2 ... yes
Checking for SDL on build host ... yes
Checking for SDL_image on build host ... yes
Checking for ALSA ... yes
Checking for PulseAudio ... yes
Checking for Linux joysticks ... yes
Checking for libsndfile ... yes
Checking for makeinfo >= 4.13 ... no
Checking for texi2pdf ... no
Checking for install ... yes
Trace mode included
Accurate audio enabled
Creating config.mak and config.h
rm -f  portalib/strcasecmp.o portalib/strsep.o becker.o breakpoint.o cart.o crc16.o crc32.o crclist.o deltados.o dragondos.o events.o fs.o hd6309.o hexs19.o input.o joystick.o keyboard.o logging.o machine.o mc6809.o mc6821.o module.o path.o printer.o romlist.o rsdos.o sam.o snapshot.o sound.o tape.o tape_cas.o ui_null.o vdg.o vdg_palette.o vdisk.o vdrive.o vo_null.o wd279x.o xconfig.o xroar.o vdg_bitmaps.o   main_unix.o portalib/glib/ghash.o portalib/glib/gmem.o portalib/glib/gslist.o portalib/glib/gstrfuncs.o gtk2/drivecontrol.o gtk2/filereq_gtk2.o gtk2/keyboard_gtk2.o gtk2/tapecontrol.o gtk2/ui_gtk2.o gtk2/vo_gtkgl.o vo_opengl.o sdl/ao_sdl.o sdl/joystick_sdl.o sdl/keyboard_sdl.o sdl/ui_sdl.o sdl/vo_sdl.o sdl/vo_sdlyuv.o sdl/vo_sdlgl.o filereq_cli.o macosx/filereq_cocoa.o macosx/ui_macosx.o alsa/ao_alsa.o oss/ao_oss.o pulseaudio/ao_pulse.o sunos/ao_sun.o macosx/ao_macosx.o jack/ao_jack.o tape_sndfile.o ao_null.o linux/joystick_linux.o windows32/ao_windows32.o windows32/common_windows32.o windows32/filereq_windows32.o windows32/xroar.res mc6809_trace.o hd6309_trace.o  xroar xroar.exe doc/xroar.info doc/xroar.pdf doc/xroar.html doc/xroar.txt tools/font2c
CC becker.o
CC breakpoint.o
CC cart.o
CC crc16.o
CC crc32.o
CC crclist.o
CC deltados.o
CC dragondos.o
CC events.o
CC fs.o
CC hd6309.o
CC hexs19.o
CC input.o
CC joystick.o
CC keyboard.o
CC logging.o
CC machine.o
CC mc6809.o
CC mc6821.o
CC module.o
CC path.o
CC printer.o
CC romlist.o
CC rsdos.o
CC sam.o
CC snapshot.o
CC sound.o
CC tape.o
CC tape_cas.o
CC ui_null.o
CC vdg.o
vdg.c:96:1: warning: variably modified ‘vdg_pixel_buffer’ at file scope [enabled by default]
CC vdg_palette.o
CC vdisk.o
CC vdrive.o
CC vo_null.o
CC wd279x.o
CC xconfig.o
CC xroar.o
CC main_unix.o
CC alsa/ao_alsa.o
CC pulseaudio/ao_pulse.o
CC tape_sndfile.o
CC ao_null.o
CC linux/joystick_linux.o
CC mc6809_trace.o
CC hd6309_trace.o
CC rasppi/ao_rasppi.o
In file included from /opt/vc/include/bcm_host.h:54:0,
                 from rasppi/ao_rasppi.c:24:
/opt/vc/include/interface/vmcs_host/vcgencmd.h:33:27: fatal error: vchost_config.h: No such file or directory
compilation terminated.
CC rasppi/keyboard_rasppi.o
Makefile:458: recipe for target 'rasppi/ao_rasppi.o' failed
make: *** [rasppi/ao_rasppi.o] Error 1
make: *** Waiting for unfinished jobs....
/home/pi/RetroPie-Setup
Could not successfully build_xroar Dragon / CoCo emulator XRoar (/home/pi/RetroPie-Setup/tmp/build/xroar/xroar not found).

Log ended at: Tue Jul 28 23:29:15 UTC 2015
Total running time: 0 hours, 1 mins, 16 secs
```

joolswills, 2015-07-29:
do you have the file `/opt/vc/include/interface/vmcs_host/linux/vchost_config.h` ? 
`

joolswills, 2015-07-29:
I can confirm the issue - no idea why it didn't happen on my dev environment but can reproduce it on my actual pi, will debug/fix.

joolswills, 2015-07-29:
ok - i added a missing include - for some reason it didn't need this in my chroot raspbian dev environment - not sure why - but it should build now.

HerbFargus, 2015-07-30:
I can verify it's building now. I'm still trying to figure out how to run the roms (I grew up after the thing was popular...) but it all seems to be running fine.

joolswills, 2015-07-30:
I shall close this then. Cheers!

closed 2015-07-30 by joolswills