RetroPie/RetroPie-Setup #914: lr-gpsp broken in RetroPie 3.0 RC1

HerbFargus, 2015-07-24:
This is the runcommand log:
```
/opt/retropie/supplementary/runcommand/runcommand.sh: line 684:  2484 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-gpsp/gpsp_libretro.so --config /opt/retropie/configs/gba/retroarch.cfg "/home/pi/RetroPie/roms/gba/Pokemon - Emerald Version (U).gba" --appendconfig /tmp/retroarch.cfg
```

My game runs fine in GPSP, but lr-GPSP acts like its loading up the rom and then it goes black and exits back to emulationstation. I tried rebuilding from binary and from source and both did not work. I made sure to have the correct bios as well. 

verified by another user here:
http://blog.petrockblock.com/forums/topic/lr-gpsp-not-working-in-retropie-3-0-rc1/

joolswills, 2015-07-24:
Run the rertoarch commandline standalone with "--verbose" for more information.

HerbFargus, 2015-07-24:
OK so- the emulator does work fine, its a configuration issue. For some reason its not reading the bios file even though the default retroarch bios directory is correct.

 To solve it I opened the rgui from the retroarch menu, navigated to settings, directory, and chose the same bios directory /home/pi//RetroPie/BIOS, changed it to save configs on exit and now lr-gpsp runs. Kinda weird.

HerbFargus, 2015-07-24:
ok after more testing the error seems to be related to support for specific games rather than the emulator itself- pretty much is described by this issue here: 

https://github.com/libretro/Lakka/issues/177 

apologies for the noise. I should be more thorough in my testing.

closed 2015-07-24 by HerbFargus