RetroPie/RetroPie-Setup #920: Can't get Qumox SNES replica to work with emulators

jry25, 2015-07-26:
Hi I bought those [Qumox SNES replica](http://www.amazon.fr/gp/product/B00PLNOXYU?psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00) but they don't work with emulators on Linux retropie 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l.

I used [the guide to configure the controller](https://github.com/RetroPie/RetroPie-Setup/wiki/RetroArch-Configuration) it worked well and the SNES replica works fine with the RetroPie interface.
But when I start a game, for example Super Mario Bros on NES, the controller doesn't work correctly. Apparently the buttons are kind of switched, I've notice that the Select is matching the D-Pad Down.

Here are my configs files, can you help me to get the controller working with the games ?
Let me know if you need more informations to help me. Thanks.

/opt/retropie/configs/all/retroarch-joypads/USBgamepad.cfg
```
input_device = "USB gamepad"
input_driver = "udev"
input_b_btn = "2"
input_y_btn = "3"
input_select_btn = "8"
input_start_btn = "9"
input_up_axis = "-1"
input_down_axis = "+1"
input_left_axis = "-0"
input_right_axis = "+0"
input_a_btn = "1"
input_x_btn = "0"
input_l_btn = "4"
input_r_btn = "5"
input_enable_hotkey_btn = "8"
input_exit_emulator_btn = "9"
input_menu_toggle_btn = "0"
input_load_state_btn = "4"
input_save_state_btn = "5"
input_reset_btn = "2"
input_state_slot_increase_axis = "+0"
input_state_slot_decrease_axis = "-0"
```

/opt/retropie/configs/nes/retroarch.cfg
```
#include "/opt/retropie/configs/all/retroarch.cfg"
# All settings made here will override the global settings for the current emulator core
input_remapping_directory = /opt/retropie/configs/nes/
video_shader = /opt/retropie/emulators/retroarch/shader/phosphor.glslp
video_shader_enable = false
video_smooth = false
```

/opt/retropie/configs/snes/retroarch.cfg
```
#include "/opt/retropie/configs/all/retroarch.cfg"
# All settings made here will override the global settings for the current emulator core
input_remapping_directory = /opt/retropie/configs/snes/
video_shader = /opt/retropie/emulators/retroarch/shader/snes_phosphor.glslp
video_shader_enable = false
video_smooth = false
```

/opt/retropie/configs/all/retroarch.cfg
```
## Skeleton config file for RetroArch

# Save all save files (*.srm) to this directory. This includes related files like .bsv, .rtc, .psrm, etc ...
# This will be overridden by explicit command line options.
# savefile_directory =

# Save all save states (*.state) to this directory.
# This will be overridden by explicit command line options.
# savestate_directory =

# If set to a directory, Content which is temporarily extracted
# will be extracted to this directory.
# extraction_directory =

[... 706 lines cut ...]

# When saving a savestate, save state index is automatically increased before
# it is saved.
# Also, when loading content, the index will be set to the highest existing index.
# There is no upper bound on the index.
# savestate_auto_index = false

# Slowmotion ratio. When slowmotion, content will slow down by factor.
# slowmotion_ratio = 3.0

# The maximum rate at which content will be run when using fast forward. (E.g. 5.0 for 60 fps content => 300 fps cap).
# RetroArch will go to sleep to ensure that the maximum rate will not be exceeded.
# Do not rely on this cap to be perfectly accurate.
# fastforward_ratio = 1.0

# Setting this to false equals no FPS cap and will override the fastforward_ratio value.
# fastforward_ratio_throttle_enable = false

# Enable stdin/network command interface.
# network_cmd_enable = false
# network_cmd_port = 55355
# stdin_cmd_enable = false

input_enable_hotkey = nul
```

joolswills, 2015-07-26:
You have posted the version of the linux kernel, no idea which version of retropie - did you start from an image?

This is more of a support request than bug and would be better posted on the retropie forum.

Doesn't look like you are running the latest 3.0 rc1 anyway - so please start with that image.

jry25, 2015-07-26:
Ok thank you I'll do that.
I didn't know where was the best place to post my issue, I was lazy and did it on GH because it was open.
I'll keep you posted if I finally manage to get the controllers work.

closed 2015-07-26 by jry25

jry25, 2015-07-26:
It worked at the first try with the 3.0 rc1. Thank you for your advice.