RetroPie/RetroPie-Setup #923: Lightguns and RetroPie

P3ter, 2015-07-28:
Hi,

Is it possible to use USB lightguns with a RetroPie setup on MAME/FBA games ?

And also, could wiimotes be used like lightguns when installed like this (method 2) : https://github.com/RetroPie/RetroPie-Setup/wiki/Wiimotes-with-classic-controllers .

Regards,
P3ter

joolswills, 2015-07-29:
If any of the emulators support using a gamepad as a lightgun to move the target, it may work with the wiimote, but I don't have an answer for you I'm afraid.

joolswills, 2015-07-29:
You might do better to see if anyone on the forum has tried - this is less of a support area, but an issue tracker for bugs/feature requests etc.

P3ter, 2015-07-29:
Ok thanks for your answer ;-)

closed 2015-07-29 by P3ter