RetroPie/RetroPie-Setup #924: bashwelcometweak preventing login at console and ssh

mrkorb, 2015-07-31:
I have a Pi 2 running Raspbian, previously setup (i.e. not a prebuilt Retropie SD card image) on it's own that I later installed Retropie 2.6 and then upgraded to 3.0 RC1.  When the bashwelcometweak is enabled, I am able to login from the console immediately following the boot sequence, but after logging out and attempting to login again, the system freezes up at the point where the bashwelcometweak should display on the screen.  Attempts to login via ssh also meet the same freeze.  My only solution is to pull the plug on the system so that I can login again following boot so that I can then disable the tweak in emulationstation and be able to login with normal function.  Reenabling the tweak results in the freeze at login once again.

I should probably clarify that when I say "freeze" the system is not completely halted, as I can ping it, access network shares, ctrl-alt-del from the keyboard works, etc etc but otherwise I'm stuck sitting there waiting for the tweak script to do whatever it is supposed to be doing before I get a bash prompt.

joolswills, 2015-07-31:
Please install from a fresh install. 2.6 is not pgradeable to 3.0 without manual work.

joolswills, 2015-07-31:
having said that, I've not seen this problem so if it is caused by one of the commands in that script, then you may have a problem with your pi - it just gets the cpu / gpu temps etc.

mrkorb, 2015-07-31:
Except that the issue presented in 2.6 before the upgrade to 3.0, so this is not something solely caused by the upgrade process, I was just describing my particular situation. Also as I said, it works just fine and shows the tweak following the first login after powering on the pi, so at least initially there is no problem until I log out and then log back in.  Is there possibly a log file somewhere that might show me what part of the script is getting hung up?  I just now tried putting the script into it's own file and running it, and that works fine when run as a standalone thing, so it's just the inclusion into .bashrc that seems to be the issue for me.

joolswills, 2015-07-31:
Did you try on a fresh 3.0rc1 install ? Not sure but maybe something was already wrong with your bashrc or similar.

joolswills, 2015-07-31:
no-one else has reported this issue, so i think it is something specific to your install.

mrkorb, 2015-07-31:
I've never knowingly altered bashrc before and it doesn't appear to have been changed by anything else.  This is running on the 2015-02-06 raspbian release which has been kept pretty basic overall.  I'll try a fresh install eventually I suppose, just no time for it now.

closed 2015-07-31 by joolswills

mrkorb, 2015-08-24:
I finally got around to looking at this closer, and found that it hangs at line 28, where it tries to get the temperature from /sys/class/thermal/thermal_zone0/temp.  As it turns out, [other](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=88127) [people](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=109938) ran into the same issue I've been having, where somewhere between 24 and 72 hours that particular command will hang and they have to pull the plug to get it working again.  [One person narrowed it down](https://www.raspberrypi.org/forums/viewtopic.php?p=734855#p734855) to there being some kind of conflict with reading the temp when they have an Adafruit LCD plate connected to the system.  I don't have that, but I do have a Piglow LED board connected and running a binary clock script on startup.  So, my best guess at this point is that having some extra doohicky attached to the GPIO (or at least attached to a particular pin) is the cause of this freeze occurring when trying to read the temperature.

joolswills, 2015-08-24:
Thanks for the info  - This should be reported upstream to perhaps https://github.com/raspberrypi/firmware or https://github.com/raspberrypi/linux/ as it is likely a kernel/firmware problem (or possibly a hardware one with your board).

mrkorb, 2015-08-28:
New [raspberrypi-bootloader](https://github.com/raspberrypi/firmware/issues/192) release appears to solve the issue.