RetroPie/RetroPie-Setup #925: Experimental Package Stability Tests

HerbFargus, 2015-07-31:
Here's the link for chart:

https://docs.google.com/spreadsheets/d/1z45RZfz7RJViKz5coGLodhtB3CnG9E_7SaHc_138qeM/edit?usp=sharing

most have been tested with a Raspberry Pi 2, I'm not sure how much the stability changes compared to a Rpi 1. I was unable to test some of them and I have a few I've yet to test- I just figure its easier to track here than on the forums.

joolswills, 2015-08-01:
Thanks - very useful. I will compare this to the module code (in some cases the module itself needs some scripting improvements even though it runs stable) and go from there.

joolswills, 2015-08-01:
I think I will move some out of experimental before 3.0 final.

Basically the stable ones that don't have any issues or need scripting improvements.

joolswills, 2015-08-01:
How would you like me to mark them as no longer in experimental - remove them to move the row to another sheet ?

HerbFargus, 2015-08-01:
You can mark them however you'd like- I can add another drop down option for "moved out of experimental" or you can toss them on another column or sheet- just as long as its clear.

joolswills, 2015-08-01:
For now I have greyed them out. feel free to improve on that :)

HerbFargus, 2015-10-15:
Pretty much resolved for the time being.

closed 2015-10-15 by HerbFargus