RetroPie/RetroPie-Setup #929: [Suggestion] Dedicated Theme/Artwork Repo

HerbFargus, 2015-08-02:
It seems as one of the things that RetroPie is lacking as far as the main code goes is with themes/artwork- I know that there are lots of splashscreens and themes littered throughout the forums and other repositories- but I think it would be nice to have a dedicated repo for everything so people know what options are available to them as well as giving them a dedicated place to share their creations. 

At the very least- as our development has passed the supported emulators in the simple theme- it would be nice to at least have the default simple theme be complete (namely adding coco, dragon 32, vidopac, sg-1000, etc.) I'd be happy to make them but I'd just like to know where I can add them to the project. I know @gizmo98 has his repo https://github.com/gizmo98/simple-theme-plus but I don't know if that's something that would be added to the retropie github group or if we would create a new repo that has the emulationstation themes and all the splashcreens or what- I'm open to new ideas.  

also if there is a new repo created I could create a simple wiki on how to build new themes as well for those who want to customise what we already have.

joolswills, 2015-08-03:
this came up before, at least in regards to the intro logos. I think we should have a repository for those and the theme on github. I'll create some repositories and move stuff there if @petrockblog and @gizmo98 are both happy with that.

petrockblog, 2015-08-03:
Good idea with the splash screens - I have created a repository . Now we can also move the splash screens from at https://github.com/RetroPie/RetroPie-Setup/tree/master/supplementary/splashscreens to the new repository. 
Regarding the EmulationStation theme I know that Nils, the author of the original ES theme "simple", has put the theme files into a Github repository at https://github.com/nilsbyte/simple. It was last updated a few months ago - should we fork that repository? Or create pull request and see how things work? What do you suggest?

petrockblog, 2015-08-03:
Thinking about it, I would suggest myself to create a dedicated repository here that can also contain all the other themes that were created bz the community and are currently published via the forum at http://blog.petrockblock.com/forums/forum/retropie-project-forum/emulation-station-2-x-themes/. Which [license](http://blog.petrockblock.com/forums/forum/retropie-project-forum/emulation-station-2-x-themes/) would you suggest to use for that repository?

taalas, 2015-08-03:
Since @HerbFargus and I have already started an repository which is used for artwork generated for the RetroPie project at https://github.com/taalas/RetroPie-Artwork 
Would it make sense to use that or incorporate the data inside if you decide to make a new repository for this?

HerbFargus, 2015-08-03:
A few thoughts:

For the splashscreen repo- perhaps it would be nice if the readme had a simple standard for creating splash screens (like the dimensions for 16:10, 16:9, and 4:3) just in case we wanted to accommodate the different types of TVs users own.

for the themes it may be useful to either fork nil's repo/ have our own repo just for the default simple theme or we could have one repo for the simple theme along with all the themes (assuming they are mostly complete) we could probably just do a simple gpl 2 or gpl 3 license and then just make sure to get permission from each user to add their theme to the repo. We could name their theme with their username as a prefix or something. 

I'm impartial either way- just as long as there is a central area to consolidate things.

(and we could also include the repo data @taalas and I have worked on as well)

joolswills, 2015-08-04:
Not sure we can enforce a licence for all 3rd party themes - that's up to the contributors - if putting all on one repository, they should all have their own licence.

for the simple theme we should fork it, so we can easily pull in from upstream. We should keep that on it's own too.

joolswills, 2015-08-06:
I was going to move the splashscreen @petrockblog but I don't have access - I thought I was an admin, but is that only for selected repositories within RetroPie or ?

HerbFargus, 2015-08-06:
I've submitted a pull request with all the splashscreens from the forum, including the ones that are already in the script- I sent private messages to all the people involved and they all were ok with me adding their splashscreens, so if @petrockblog merges my pull request we can just delete the current folder here.

joolswills, 2015-08-06:
I missed that - nice one can't merge it though due to my permissions :(

HerbFargus, 2015-08-06:
Did we come to a consensus on the themes? did we want it all together or did we want to have a fork of the default simple theme and then a separate repo for all the user created themes (I can think of 3 off the top of my head that would be considered complete enough to add to our repo- it would also resolved two issues on this repo as well.)

joolswills, 2015-08-06:
The themes are big, so I def want the main simple theme on it's on on a fork (for ease of maintenance and to keep base installs down). As for the others they can be maintained in a single repo I guess, and we can have it so users can install them manually or via a module etc.

joolswills, 2015-08-06:
I've forked the simple them here - https://github.com/RetroPie/simple I'll adjust the install script to use it. Feel free to submit pull requests for empty platforms etc everyone ;-)

joolswills, 2015-08-06:
s/empty/missing

HerbFargus, 2015-08-06:
That sounds reasonable to me- I also agree with the separate licenses- each creator can just add licenses within their own theme folders. I think it would be good to have a completion stipulation (I don't care much for themes that have only a few systems.)

joolswills, 2015-08-06:
I really should have named the repository better though eh. I'll rename it.

robertybob, 2015-08-08:
Forgive my ignorance, but once merged, how would we install new themes? Would it be an option in the RetroPie menu in ES? Could there be some kind of thumbnail/preview so we know what to expect before installing it?

joolswills, 2015-08-09:
it will install themes initially and work exactly as before. later it may be  improved.

joolswills, 2015-08-10:
themes and splashscreens are now moved. closing.

closed 2015-08-10 by joolswills

HerbFargus, 2015-08-10:
Well the simple theme and splashscreens have been moved, but we still haven't integrated user submitted themes (there are a couple nice themes like zoid, static canela, and nbba)

joolswills, 2015-08-10:
Perhaps we can have a new ticket - we have far too many tickets that cover multiple topics and it gets harder to manage as I have to read through each time rather than going from the title etc. :)