RetroPie/RetroPie-Setup #930: Freezing controls

amm297, 2015-08-03:
I have two poblems one is that if I strat a game and don't play it for more than 10 minutes the control freezing and not work,  but the keyboars still working, as controls I'm using a classic retroarch machine 

and the second is that if play long time the game the rpi star to make rare things, i don't know how to explain it

joolswills, 2015-08-03:
what version ? which emulator(s) ?

amm297, 2015-08-03:
version 3.0 retropie, and advmame emulator

joolswills, 2015-08-03:
3.0rc1 ? have you tried it with another controller ?

amm297, 2015-08-03:
i'm using the latest version of retropie and no I didn't try another controller because i'm using an old retroarch machine and I can't change the controllers

joolswills, 2015-08-03:
nothing has changed with advmame for some time, so I'm not sure what the problem could be - and it only happens with advmame ? nothing else has changed on your set up ?

joolswills, 2015-08-03:
what do you mean by "old retroarch machine" ? advmame is unrelated to retroarch btw

amm297, 2015-08-03:
http://www.vadejuegos.com/imagenes/2010/09/28/zonaarcade.jpg machime like that, I don't know the name in English
I only use the advmame, the rest of emulator there haven't interest for mi

joolswills, 2015-08-03:
to rule out something hardware related, it would be useful if you can get a usb controller from somewhere, and connect that just for debugging/testing. Also if you try a retropie 2.6 image, does that work ? Does it happen on both versions of advmame (retropie 3.0 has two versions switchable in the pre load menu)

amm297, 2015-08-03:
The problem star when I install this emulator on rpi https://github.com/hissorii/px68k, could be possible?

joolswills, 2015-08-03:
not sure but as that doesn't come with retropie it is worth testing from a fresh image.

amm297, 2015-08-03:
Ok, thanks I'll try on fresh image

amm297, 2015-08-11:
Yes, I reinstall and it seems to fixed it, maybe I did some worng at last installation

closed 2015-08-11 by joolswills