RetroPie/RetroPie-Setup #931: Multiple players on single player games

arthurlutz, 2015-08-04:
This is a far fetched issue I'm submitting, but it's a way to start discussing this idea.

The idea would be to enable a multiplayer mode for single player games where players would play side by side the same game. For example a 2 player launch would launch two windows side by side with an instance of retroarch configured for player1 on the left and an instance of retroarch configured for player2 on the right. The players would then manually synchronize and play together (for speed or points, or anything the imagination can think of). 4 players would split the screen in four. With the size of flat screen TVs and the small resolution of retro games, this would be visually feasible right ?

Technically you could use a lightweight programmable window manager to spawn 2 or 4 windows of retroarch (dwm http://dwm.suckless.org/, i3 http://i3wm.org/, awesome http://awesome.naquadah.org/, etc.).

joolswills, 2015-08-04:
This would require massive changes to retroarch, and it is out of the scope of retropie. I very much doubt for a variety of technical reasons it will ever happen - so I'm not sure there is any point discussing it here sorry.

joolswills, 2015-08-04:
I re-read again what you wrote and I see that you are thinking to just run multiple copies of retroarch -  - i misread the first time - but it still doesn't sound like it would work to me -but it's definitely not something I would have interest in looking at.

joolswills, 2015-08-04:
(But if you are interested in such a thing, why not try it yourself ?)

HerbFargus, 2015-08-04:
You can sorta do this with tgbdual- but not with any of the other emulators that I know of as of right now.

arthurlutz, 2015-08-04:
@HerbFargus thanks for the tip. This is a more "advanced" use case where the games know how to talk to each other through a protocol (in your example serial link emulated over tcp). It's pretty cool and there seems to be a libretro port for it https://github.com/libretro/tgbdual-libretro 

I'm looking for a more basic approach : run two games in parallel without communication but separate controllers.

joolswills, 2015-08-04:
Try it - build retroarch for pi with Xserver support, and test out running multiple copies with different configs ?

It's not something that will be integrated into retropie though - and we don't currently build retroarch with xserver support.

KevinGage, 2015-08-04:
You might be able to get multiple instances started using screen.  
http://unix.stackexchange.com/questions/7453/how-to-split-the-terminal-into-more-than-one-view

I think the hard part will be sending input to both instances at the same time.  One will likely have focus while the other does not.
Picture having two instances of notepad open in windows.  When you type it only shows up on one at a time.

jedahan, 2015-08-04:
maybe spawning a second xserver, configured to work only with the second gamepad...

closed 2015-10-10 by joolswills