RetroPie/RetroPie-Setup #932: No Video output on lr-handy

joolswills, 2015-08-04:
Reported here:

http://blog.petrockblock.com/forums/topic/lynx-no-video-output-v3-0-rc1/

Bisected and reported upstream. 

https://github.com/libretro/libretro-handy/issues/9

joolswills, 2015-08-04:
I debugged the problem and made a pull request for upstream with the fix.

HerbFargus, 2015-08-04:
I rebuilt from source and I can confirm it is now working.

joolswills, 2015-08-04:
I have updated the other ticket. feel free to read. I'll close this once I have built new binaries.

HerbFargus, 2015-08-04:
Well I'll say thanks regardless of your code format :D

joolswills, 2015-08-05:
Cheers. well the point is mixing indentation in a single file becomes a mess, and it's pretty common in development not to do it. But when there is ego involved. well..

joolswills, 2015-08-05:
New binaries. built.

closed 2015-08-05 by joolswills