RetroPie/RetroPie-Setup #933: Changing some retroarch defaults

joolswills, 2015-08-04:
@HerbFargus @petrockblog @gizmo98 

feedback welcome.

I would like to change some default retroarch settings. I would like to turn off video_threaded by default, as it is detrimental to perfect video sync.

https://github.com/libretro/RetroArch/wiki/Getting-optimal-vsync-performance

>Threaded video fallback

>If your video driver has very bad performance, it is possible to run it on a thread to avoid almost all video driver overhead. Set video_threaded = true in config. Butter smooth VSync behavior in this case is impossible however, and latency might increase slightly. Use only if you cannot obtain full speed otherwise.

I think in the majority of cases it is better off. Most emulators are full speed on the pi1 with it off also from my testing. Or we could turn it off just for the pi2 if you think there are pi1 emulators that benefit from the setting.

Also I was thinking about pixel shaders. By default we have video_smooth off, and "some" emulators have some pixel shader configs in the system specific retroarch.cfg. Some users just want video smoothing on, and it's a pain for them as they need to edit lots of files. What would be easier would be to put the shader configs in, but left disabled, and those that want shaders can configure/enable - that way the core retroarch video_smooth option can be easily changed.

What we have currently for example is /opt/retropie/configs/gb/retroarch.cfg

```
input_remapping_directory = /opt/retropie/configs/gb/
video_smooth = false
video_shader = /opt/retropie/emulators/retroarch/shader/hq4x.glslp
video_shader_enable = false
```

What I propose 

```
input_remapping_directory = /opt/retropie/configs/gb/
# video_smooth = false
video_shader = /opt/retropie/emulators/retroarch/shader/hq4x.glslp
# video_shader_enable = true
```

so the user can easily enable the lines above if they want to use a shader. we could easily have a simple gui to then configure video_smooth on or off.

I prefer video_smooth on by default, if I'm lazy to configure shaders, as on modern LCD's without everything looks very blocky. but whatever the default is, it should be easy to switch.

joolswills, 2015-08-04:
My personal preference out of the box would be video smooth on and shaders off. then have a retroarch configuration tool to allow editing of core retroarch.cfg and have some presets also. but im happy to discuss the defaults. I think the scaling artifacts are more visible with video smooth off also.

HerbFargus, 2015-08-04:
I've only really tested the settings with my PC- the video threaded was necessary on my pc to get things to work without performance/audio lag, but if you've tested it on the pi and it runs fine without then I have no qualms. To be honest I've hardly touched retroarch configs minus a few core options and testing out remapping configs. @biscuits99 may have better input on configs as he seems to be a lot more involved on that front.

joolswills, 2015-08-04:
it was a while ago I tested without video threading, so I'd be interested in any feedback. If I make a little tool for retroarch configuration, I would include this as an option of course so it would be easily switchable.

HerbFargus, 2015-08-04:
If I've got time later today I'll see if I can run some tests with the configurations you've proposed. Is this something that would be integrated into the runcommand menu or would the tool be its own thing in the setup script?

joolswills, 2015-08-04:
thanks.

well, for retroarch core settings I was thinking a retropie-setup module, but runcommand could perhaps support editing of per core settings (definitely leaving that for 3.1 though). but was thinking to get the ideas on here done for 3.0 final.... if people agree and they work out.

biscuits99, 2015-08-04:
Maybe this is a step far too far at the moment, but I've been working on this to make it very easy for users to update their video output from the default:
https://github.com/biscuits99/rp-video-manager

(I use the #1 "overlays" option under the "Overlays for 1080p" menu.) 

Its not well written, but at a press of a button, you get CRT like output with overlays. Appreciate this may be overkill. Its main disadvantage is that I'm not at all sure how to write it so it inserts lines into an existing config as opposed to overwriting it (I presume I need to use "sed")

In answer to the original question, the proposal looks good as it leaves the default setting with it turned off, as opposed to forcing anything on the user, and it also makes it easy to enable a basic shader or smoothing.

As an aside, I've added a very popular default shader here.
http://pastebin.com/raw.php?i=bhVvvE8w
(crt-hyllian-sharpness-hack.glsl)

HerbFargus, 2015-08-04:
Turning off video_threaded had no noticeable performance issues on my rpi2- I tested with a few emulators (picodrive, snes9xnext, handy) I would assume the results will be similar for the others. 

Also had no noticeable issues with the settings you proposed.

joolswills, 2015-08-05:
@biscuits99 looks good - will try it out.

joolswills, 2015-08-05:
I wonder if we should clean up our shaders, and install just a small selection by default, and have the ability to install more - There are plenty we have currently, that are not that useful.

biscuits99, 2015-08-05:
I like that idea. I could run through them and name the "pointless" ones that are more novelty than anything else.

joolswills, 2015-08-05:
yeh that would be useful thanks.

joolswills, 2015-08-05:
also any that have horrible performance can probably be removed by default.

biscuits99, 2015-08-05:
Didnt realise there are so many in sub folders.
Here is my view on the root files
http://pastebin.com/raw.php?i=4SEsAFMc

I'd suggest someone checking the ones I've marked as ok, as I may be too lenient. Tested on a Pi2.

joolswills, 2015-08-06:
The ones in subfolders also can't be switched to automatically with prev/next. based on your list - thats for that - we can def clean up the root, and decide if any others should be added to the root also, to make them more accessible.

joolswills, 2015-08-10:
The defaults have been changed. We will find out from the number of complaints if people don't like it - but at least it's easy to switch.

joolswills, 2015-08-10:
Regarding the shader stuff - we should perhaps open a new ticket for that.

closed 2015-08-10 by joolswills