RetroPie/RetroPie-Setup #936: Upcoming Releases Wishlist

HerbFargus, 2015-08-07:
This is a doc of potential viable improvements for the future releases of RetroPie

https://docs.google.com/spreadsheets/d/1NXG3px6huJ7JzKcqaFVoaeXJ34jfuvpxIcCLGvNfXNg/edit?usp=sharing

joolswills, 2015-08-07:
Looks good. thanks! I think #1 will be fixed automagically in the coming weeks.

HerbFargus, 2015-08-07:
I'm hoping that's the case, though I still think its compiled without the joypad support enabled, so we may need to sort that out.

ProxyCell, 2015-08-07:
would it be inappropriate to request that jstest be installed? (apt-get install joystick)
it is an extremely useful package that lets you test your buttons/keys for input devices on the command line

i would write a script that launches it, and closes it after say 1-2 minutes automatically (no need to kill it with a keyboard)

joolswills, 2015-08-07:
jstest is already installed on the retropie images. if you want a module to make sure it's installed it can be added to the dependencies for the module. I'll consider something like that for a future release.

ProxyCell, 2015-08-07:
@joolswills - thanks a lot, yeah i would like it there :)

gonzo-the-great, 2015-08-12:
hi would it be possible to add this to the list,

support for gamelist.xml in sub rom folders, so that roms can be split into genre/region and still have the functionality of the awesome gamelist.xml feature

Death259, 2015-08-14:
FYI, I'm working on #8 on the list. @joolswills how would you like me to get the code to you? Do you want me to branch and then issue a pull request, or just copy and paste the code into a new issue? The code might not quite be up to snuff with your style guide, so you might have to either make some changes or tell me what needs to be changed.

joolswills, 2015-08-14:
thanks. can submit a pull request if you like, or if you wanted some comments before feel free to use issue tracker. I

ProxyCell, 2015-10-08:
can i still add to this?
there is an item on the list now for "cloud support" of roms
i want to mention that cloud support of other things like customized preferences, save games, save states and even the gamedata (the xml files with the names, publishers, etc, box art) is possibly of greater importance than rom files are

joolswills, 2015-10-08:
That seems a bit out of the scope of the project - if you need this you could try something like https://github.com/joe42/CloudFusion

joolswills, 2015-10-08:
to be honest. instead of having a ticket here pointing to a spreadsheet, feature requests should just be individual tickets - but they may well be closed if they seem a bit far fetched.

ProxyCell, 2015-10-08:
thats fine, i prefer simpler systems that rely less on cloud/network access - but what about the rom cloud thing? is that still kosher with the project's scope then?

joolswills, 2015-10-08:
I have no interest in adding it - you can always sync your rom folder yourself or try a cloud based fuse filesystem etc.

ProxyCell, 2015-10-08:
no problemo, wasn't my suggestion :)
just thought it was a bit weird to see

pancakeykakes, 2015-10-08:
Upvote for  #1, #4, #5, and #6.

joolswills, 2015-10-08:
wasn't mine either. Maybe it's Herbs or someone who mentioned it to him. anyway, we should close this and add the features to tickets.

adrianmoisey, 2015-10-08:
> i want to mention that cloud support of other things like customized preferences, save games, save states and even the gamedata (the xml files with the names, publishers, etc, box art) is possibly of greater importance than rom files are

Not quite what you're after, but I've written some scripts that will install retropie for me (with only the consoles that I want), configure my gamepad and then copy all my roms (and gamelists and cover art, all generated by scraper) to it.
If I want a fresh install, I can rerun my scripts and have a new pi all setup and ready to go.
I haven't done anything like getting saved stated off of it yet, but that's less important to me now.

joolswills, 2015-10-08:
When I say about opening ticket - tickets should be opened for ones that actually may happen - I see the usefulness of floating some ideas around first. No point opening a ticket for something that will never be done :)

HerbFargus, 2015-10-08:
This was more a brainstorming thing for future ideas but I'll stick the ideas for future enhancements on the forum  and people can submit ideas as individual tickets like the tracker should be used. Closing now.

closed 2015-10-08 by HerbFargus

ProxyCell, 2015-10-08:
@adrianmoisey - i would be very interested in seeing your scripts that relate to retropie and the other ones you mentioned - would you be willing to contact me outside of github? proxycell@iruberleet.org

Death259, 2015-10-08:
I can tell you that the cloud idea came from my [PiAssist](https://github.com/Death259/PiAssist) application. I've got it setup so that you can upload/download save files and save states from Dropbox.

adrianmoisey, 2015-10-09:
> @adrianmoisey - i would be very interested in seeing your scripts that relate to retropie and the other ones you mentioned - would you be willing to contact me outside of github?

I plan to stick it on github soon. I'll ping this issue when I do