RetroPie/RetroPie-Setup #943: User Submitted Theme Repo

HerbFargus, 2015-08-10:
A centralised repo for extra theme options would address these 3 issues- a few from the forum could probably also be added as well.

https://github.com/RetroPie/RetroPie-Setup/issues/717

https://github.com/RetroPie/RetroPie-Setup/issues/616

https://github.com/RetroPie/RetroPie-Setup/issues/858

http://blog.petrockblock.com/forums/topic/nothing-but-box-art-wip/

HerbFargus, 2015-08-11:
Once it's created I'd be happy to add them to the repo setupscript similar to https://github.com/RetroPie/RetroPie-Setup/blob/ce8ede4defe4d2338255a202eb8e7b69be5bd7b4/scriptmodules/supplementary/emulationstation_theme_colorpi.sh.

petrockblog, 2015-08-11:
Would you suggest to have individual repositories for each of the themes?

HerbFargus, 2015-08-11:
I think it would be easier to have one repo for all user submitted themes and each theme could have its own license.

joolswills, 2015-08-11:
yeh I think one repo each would be good. we can manually add them into a theme installer module.

petrockblog, 2015-08-12:
I have created a repository for user created ES themes now. Initially, I have added the first three themes mentioned above. The fourth one (hosted at Mega) is not accessible at the moment. But that theme also looks very interesting ... 

I have simply put the files of each theme into their own theme subdirectory - nothing more. Now, a script module needs to be written and maybe some other things regarding the installation of the theme files. I hope this is a good starting point.

joolswills, 2015-08-12:
As discussed earlier it would be better for one repo per theme - else to install a single theme, we need to checkout the entire repository - themes can grow to a reasonable size, so it would be better to have them separate.

petrockblog, 2015-08-12:
Uups, I misread your last post. Sure thing, I will change this right now.

joolswills, 2015-08-12:
:) sorry - I could have worded it better.

joolswills, 2015-08-12:
es-theme-NAME perhaps ?

petrockblog, 2015-08-12:
Yes - I have also thought of exactly the same naming schema.

HerbFargus, 2015-08-12:
Looks good to me! Here's a link to the NBBA theme if mega isn't working for you

https://www.dropbox.com/s/sw6m0gxz22s7lz4/nbba_theme_v1_1.zip?dl=0

petrockblog, 2015-08-12:
All right. I have created a repository for each of the themes now: https://github.com/RetroPie. I also prefer the approach "one repository per theme".

HerbFargus, 2015-08-12:
Sounds good- are we also still going to keep color pi? https://github.com/RetroPie/RetroPie-Setup/blob/ce8ede4defe4d2338255a202eb8e7b69be5bd7b4/scriptmodules/supplementary/emulationstation_theme_colorpi.sh

petrockblog, 2015-08-12:
Good point! Missed that one. But I have just added it also as new repository.

robertybob, 2015-08-12:
There is/was an issue with the Simplified Static Canela theme where it [wouldn't appear properly when viewed in 4:3 ratio](http://blog.petrockblock.com/forums/topic/theme-simplified-static-canela-v1-0a/page/2/). I don't know whether this was corrected by @InsecureSpike before he raised #858 but someone may need to check that :) Also, he was planning on bringing out a series of 5/6 themes (including a new version of this particular theme).. he hasn't signed in to the forums for a while now but I tested them all and they look great ! I hope he logs on again soon as they'd be brilliant additions.

Also, as I posted on #616 , I don't think the Zoid theme doesn't come with any artwork for the RetroPie module/menu (?) Correct me if I'm wrong :)

HerbFargus, 2015-08-12:
There isn't a retropie menu option in the zoid theme yet but now that they are on their own repos I'm sure you could submit a pull request with a solution :) https://github.com/RetroPie/es-theme-zoid

robertybob, 2015-08-12:
Sorry @HerbFargus I have absolutely no idea how use Github in that way :)  [here's a link to my hastily knocked up one] (https://www.dropbox.com/s/uox4j9d2lkyua2p/retropie.zip?dl=0)

HerbFargus, 2015-08-12:
All the more reason to learn :) I'll give it a test and see i there's anything i need to clean up (I would like to have all the themes have svg logo if at all possible) Once it looks good, I'll submit a pull request for it and I'll detail what I did so that in the future you can also contribute your changes.

HerbFargus, 2015-08-12:
I traced the retropie bitmap and created an svg from it and fixed the xml as it was linked wrong to the logo as well as the xml for the whole theme (whoever created the zoid theme didn't change all their internal xml's to the zoid xml so they are all still pointing to simple.xml. so I renamed zoid.xml to simple.xml and it fixed it)

Here's a quick how to on creating pull requests- this is using the windows program but you can also use powershell. There are also plenty of other resources for learning github as well online.

### How To create a pull request

access the repo you wish to add to and fork it- so in this case:
https://github.com/RetroPie/es-theme-zoid
click fork on the top right
Then if you prefer a windows interface you can install github for windows and click the plus sign on the top left and click clone and select the repo you just forked
Press ok and it will be installed by default in the github folder in the documents folder
Then you can create a new branch by clicking the branch thing just to the left of master on the top left and in my case I called it retropie-folder
Then I just added the files locally in the aforementioned github folder in documents
Then back in github it shows my commits I've made on the branch I created and I write a short summary of the commits, press commit at the bottom and then click publish on the top right.
Then on https://github.com/RetroPie/es-theme-zoid it shows a green compare and pull request button- click that and then click create pull request. Done and done :)

robertybob, 2015-08-13:
Cheers for that!!! :) On the topic of themes ..I don't quite understand the CC licence - can anyone alter a theme created under CC and do with it as they wish, so long as they credit the original author? 

I ask as there are a couple of themes I've found for Recalbox which I've tried on my Windows copy of ES , and they look pretty good, but may need alterations made to be compatible with RetroPie. Would I *have* to ask the original author before making any adjustments and submitting here (for example), or could I get away with just crediting the author?

HerbFargus, 2015-08-13:
seems it goes both ways: https://github.com/digitalLumberjack/recalbox-os/issues/107 ha ha- 

They are multiple varyations of the cc license so It's always good to ask though just in case. Not sure if the retropie project will want to look like recalbox though as they are two separate operating systems.

joolswills, 2015-08-13:
Can this be closed now ? Any theme specific issues can be opened on on corresponding repositories. Unless I've missed any other bits of this ticket ?

HerbFargus, 2015-08-13:
I think it can be closed.

joolswills, 2015-08-13:
Cheers.

closed 2015-08-13 by joolswills