RetroPie/RetroPie-Setup #944: GPSP

expandables2, 2015-08-11:
Hi why you choose this gpsp https://github.com/gizmo98/gpsp  over libretro gpsp https://github.com/libretro/gpsp

HerbFargus, 2015-08-11:
The latest retropie build has lr-gpsp as the default though I'm quite partial to lr-mgba (though it may not perform as well on a rpi-1)  for example pokemon emerald doesn't run in lr-gpsp but it runs in lr-mgba and gpsp. The direct port may have better performance but lr-gpsp has the convenience of retroarch configs, but you can easily change the default emulator as a rom loads from the runcommand menu if the default emulator is not to your liking. 

By the way in regards to that whole omxplayer ordeal we've got it sorted and it will be integrated into the retropie 3.0 stable release which will be released any day now.

joolswills, 2015-08-13:
Yeh we don't choose gpsp over the libretro one - we include both with lr-gpsp as default

closed 2015-08-13 by joolswills