RetroPie/RetroPie-Setup #945: Add Ssel

HerbFargus, 2015-08-11:


joolswills, 2015-08-11:
A good start - but will need some work to comply with our styleguide, and install folders should be part of /opt/retropie etc if possible. There are other fixes needing in regards to hardcoded paths/users etc

HerbFargus, 2015-08-11:
I haven't tested it outside of /usr/local/bin but I'll test it out a bit later- I don't see any reason why it shouldnt work in opt/retropie

joolswills, 2015-08-11:
I would prefer that we don't use code from the other project - partly because it doesn't fit with our styleguide, but also, because it was made as a separate project for a reason I guess - rather than contributing the stuff directly to this project. Also although we have a compatible licence our code requires everything gets copyrighted to the project also, which will likely conflict.

HerbFargus, 2015-08-11:
that's a fair point- guess I need to work on my coding skills so I don't need to use other's projects as a crutch :P

ProxyCell, 2015-08-13:
Can't this bash script just be ripped apart and formed to the RetroPie standards? I feel this is a HUGE improvement for the project.

joolswills, 2015-08-13:
Well - It was already planned to implement this so don't panic - I've had it in mind for some time, so it will be done. It's not just a matter of our styleguide though regarding the above script - but I will write a scraper module when I have done with current tasks - All contributions are appreciated of course even if not accepted right away.

robertybob, 2015-08-13:
@joolswills is there a particular reason this can't be integrated into Emulationstation to replace the default scraper that so many people still use and struggle with? If it's a case that the scraper is hard coded and you're not able to make any changes to it then that's fair enough :)

joolswills, 2015-08-13:
Well work was done to integrate it, but then Aloshi got busy with real life.

I don't fancy maintaining large custom ES changes right now. Hopefully ES will become less fragmented in the future and be continued.

Death259, 2015-08-14:
FYI, feel free to rip apart my code and add it to the official retropie code. I started my [PiAssist](https://github.com/Death259/PiAssist) project to fill some gaps that I thought existed. I didn't want to add it to RetroPie at the time, because I felt that my code would be far inferior to what you guys would want. After having worked with bash enough and seeing how you guys do things, I am going to start re-writing parts of my project so that they can be 100% integrated in RetorPie. Hopefully I can get things mostly there so that you guys won't have to do as much work to make it match your style. I've given Herbfargus the go ahead to do that as well. No since in reinventing the wheel, so have at it.

Death259, 2015-08-16:
Here's my attempt at the scraper. It uses `$rootdir`, so it should work in the future if the main directory changes. I've made sure that the variable/function names and use of square brackets match what is requested in the style guide.

    #!/usr/bin/env bash

    # This file is part of RetroPie.
    # 
    # (c) Copyright 2012-2015  Florian Müller (contact@petrockblock.com)
    # 
    # See the LICENSE.md file at the top-level directory of this distribution and 
    # at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
    #

    rp_module_id="scraper"
    rp_module_desc="Steven Selph's Scraper"
    rp_module_menus="3+"
    rp_module_flags="nobin"

    function install_update_scraper() {    
        local scraper_version=$(wget -qO- https://api.github.com/repos/sselph/scraper/releases/latest | grep tag_name | sed -e 's/.*"tag_name": "\(.*\)".*/\1/')
        local scraper_location="$rootdir/supplementary/scraper"

        [[ ! -d "$scraper_location" ]] && mkdir "$scraper_location"

        #if scraper doesn't exist then download it
        local update_scraper=false
        if [[ -f "$scraper_location/scraper" ]] ; then
            local old_version=$("$scraper_location/scraper" -version 2> /dev/null)
            if [[ $? -ne 0 ]] || [[ "$scraper_version" != "$old_version" ]] ; then
                if (dialog --title "Update Scraper?" --yesno "Would you like to update the scraper script?" 0 0) then
                    update_scraper=true
                fi
            fi
        fi

        if [[ ! -f "$scraper_location/scraper" ]] || [[ $update_scraper == true ]] ; then
            #if raspberrypi1 then download the build for raspberrypi1
            if grep -q ARMv7 /proc/cpuinfo ; then
                wget https://github.com/sselph/scraper/releases/download/$scraper_version/scraper_rpi2.zip -q -O "$scraper_location/../scraper_rpi2.zip"
                unzip -qq -o "$scraper_location/../scraper_rpi2.zip" scraper -d "$scraper_location"
                rm "$scraper_location/../scraper_rpi2.zip"
            else
                wget https://github.com/sselph/scraper/releases/download/$scraper_version/scraper_rpi.zip -q -O "$scraper_location/../scraper_rpi.zip"
                unzip -qq -o "$scraper_location/../scraper_rpi.zip" scraper -d "$scraper_location"
                rm "$scraper_location/../scraper_rpi.zip"
            fi
        fi
    }

    function scrape_all_roms_scraper() {
        local scraper_location="$rootdir/supplementary/scraper"
        
        install_update_scraper
    
        $scraper_location/scraper -scrape_all -thumb_only
    }

    function scrape_specific_systems_scraper() {
        local scraper_location="$rootdir/supplementary/scraper"
        install_update_scraper
    
        local system_names=() 
        local system_name 
        local paths=() 
        local path
        local options=()
        i=0
        while read system_name; read path; do 
            system_names+=("$system_name") 
            paths+=("$path") 
            options+=("$i" "$system_name" OFF)
[... 36 lines cut ...]
        fi
    }

    function sselph_rom_scraper() { 
        while true; do
            local cmd=(dialog --backtitle "$__backtitle" --menu "Scraper by Steven Selph" 22 76 16) 
            local options=( 
                1 "Scrape All Systems" 
                2 "Scrape Specific Systems" 
            ) 
            local choice=$("${cmd[@]}" "${options[@]}" 2>&1 >/dev/tty) 
            if [[ -n "$choice" ]]; then 
                case $choice in 
                    1) 
                        scrape_all_roms_scraper 
                        ;; 
                    2) 
                        scrape_specific_systems_scraper 
                        ;; 
                esac 
            else 
                break 
            fi 
        done 
    }

joolswills, 2015-08-16:
Thanks but this is going to more work before it can be accepted. 

please use "$md_inst" instead of $rootdir/supplementary/scraper - you dont need another variable for it

your variable names are a little overly verbose. gamelist_folder_name could just be folder, as the context is already clear from the function/loop

there are camelcase vars that should be changed

there are hardcoded paths like /home/pi which should be $home or hardcoded user pi which should be $user.

Where is get_rom_information_from_es_systems_config defined ? this function name should be shortened.

regarding the bug mentioned with the xml  tag - has this been reported upstream ?

Ideally the module would build from source, and not use 3rd party binaries. at least so we can easily rebuild new binaries etc (in the same way as emulator modules or emulationstation)

joolswills, 2015-08-17:
Regarding source, the meantime if installing binaries, it should use the "isPlatform" call to check which platform

```
if isPlatform "rpi1"; then
if isPlatform "rpi2"; then
```

joolswills, 2015-08-28:
I'm currently working on this btw - started off with the above code, but looks like I will rewrite most of it.

Making an assumption that roms are in ~/RetroPie/roms too - instead of parsing gameslists (when we possibly move to ES using sqlite3 I don't want to rewrite it). If someone has put their roms elsewhere and modified ES configs, they will need to run it manually. Keeps it simple then.

HerbFargus, 2015-08-28:
Thanks @joolswills I was mucking with it last night and could only get the scrape all function to work (which sselph's just incorporated mame into) much appreciated.

joolswills, 2015-08-29:
What is the preferable location for gamelists/images ? in the rom folders or in .emulationstation ? by default the scraper would use each rom folder, so maybe that is best?

HerbFargus, 2015-08-29:
Yeah sselph's scraper keeps the images and gamelists in their respective rom folders

robertybob, 2015-08-29:
Could it be changed to Emulationstation? The systems that cannot currently by scraped will have their gamelists and images (from the built in scraper) there anyway. May get confusing having them in two different places.

joolswills, 2015-08-29:
but easier to backup lists and roms if they live with the roms themselves. but maybe I'll make it an option.

HerbFargus, 2015-08-29:
yeah it does make it easier to back up in each rom folder, but options are a good idea.

joolswills, 2015-08-29:
Interested in hearing peoples preferred configurations for the scraper, such as if they use thumb-only, or prefer screenshots over marquees etc (please give me your commandline params)

sselph, 2015-08-29:
This is great!

The reason to use the binary instead of source is because debian/wheezy has a very old version of Go in it's repository and it is unable to compile my code. You could use the unofficial binary of Go at http://dave.cheney.net/unofficial-arm-tarballs which would allow you to compile. It would require a ~50MB upfront download but would make upgrading fast and use very little bandwidth.

joolswills, 2015-08-29:
thanks - and thanks for your work. This probably should have been added a long time ago, but then there was some hope that your scraper code would make it into ES, but unfortunately Aloshi is very busy with real life it seem.

I'll probably just wait until we move to jessie, and then add an option to build from source, but I'm quite ok using your binaries for now :-)

joolswills, 2015-08-29:
(I know there is an ES branch with your code in - what stage did it get to ?) ? I'd be happy to take in pull requests against our ES fork for retropie.

sselph, 2015-08-29:
The code is in the unstable branch of ES. It is functional but I haven't been keeping it to the same level as my code since my first pull request to make a fix has been there since March. It is missing PSX, Dreamcast, and ScummVM support which was added later. It is missing MAME support which in my code is just rip off of a PR from ES.

sselph, 2015-08-29:
I forgot mention the UX could be improved. There is a third scraper now "TheGamesDB w/SHA1" and to make it work well you turn off the "user decides on conflict" otherwise you have to hit skip when there isn't a match. That was something Aloshi wanted to fix where you have multiple scrapers chosen and it would work through a progression like a platform specific scraper then do an automatic scraper then a manual scraper.

I think it is also slightly slower since it is updating UI elements as it progresses vs printing a log entry.

FlyingPikachu, 2016-02-06:
Man, this sure would be nice!

joolswills, 2016-02-17:
scraper module now has ability to enable/disable thumb-only - and it has been moved out of the experimental menu

closed 2016-02-17 by joolswills

sselph, 2016-02-17:
Just FYI, not using -thumb_only and using -workers 4 might lead to the raspberry pi running out of memory or cause paging. Each worker uses ~125 MB to process the full sized images since it decompresses the image then recompresses. More detail in this issue https://github.com/sselph/scraper/issues/61

joolswills, 2016-02-17:
thanks. We can document it or reduce workers perhaps depending on free ram - should be ok on a 1GB ram machines though.

joolswills, 2016-02-17:
By default it is set to use "-thumb_only" also.