RetroPie/RetroPie-Setup #946: lr-snes9x-next segmentation fault

raedeus, 2015-08-12:
Greetings wonderful, helpful community of retro-gamers. I need some help.

After some time fiddling with this great setup script on a clean rapbian system, I have run into a puzzling issue. I'll get right to it:

When I try and launch a particular SNES rom (in this case, Donkey Kong Country) in both .zip and .smc format I get an interesting error. I am pointed to /tmp/runcommand.log and get this:

PPU.RenderSub = 0
PPU.FullClipping = 1
Settings.Transparency = 1
Settings.SpeedhackGameID = 2
PPU.SFXSpeedupHack = 0
coldata_update_screen = 1
/opt/retropie/supplementary/runcommand/runcommand.sh: line 696:  4813 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x-next/snes9x_next_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/Donkey Kong Country.smc" --appendconfig /tmp/retroarch.cfg

I have tried other SNES roms, not a lot, but some, and they work. I thought for a second it was because of spaces, but, other rom file names contained spaces and ran fine (Earthworm Jim.zip, Super Metroid.zip, etc.).

Please forgive me if I have done any of the following with this:

1. Repeated a question regarding a known bug
2. Did something wrong in my own config (I haven't configured much)

Any help and/or input would be greatly appreciated. I am at a loss.

Cheers.

Avatar2200, 2015-08-12:
I also have those segmentation faults. About 30% of all snes games seems to be affected on my rpi2 v3.0 final:

PPU.RenderSub = 0
PPU.FullClipping = 1
Settings.Transparency = 1
Settings.SpeedhackGameID = 0
PPU.SFXSpeedupHack = 0
coldata_update_screen = 1
/opt/retropie/supplementary/runcommand/runcommand.sh: line 696:  3021 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x-next/snes9x_next_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/Super Mario Kart (E) [!].smc" --appendconfig /tmp/retroarch.cfg

video_refresh_rate = 60
video_fullscreen_x = 640
video_fullscreen_y = 480

&

PPU.RenderSub = 1
PPU.FullClipping = 1
Settings.Transparency = 1
Settings.SpeedhackGameID = 0
PPU.SFXSpeedupHack = 0
coldata_update_screen = 1
/opt/retropie/supplementary/runcommand/runcommand.sh: line 696:  2713 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x-next/snes9x_next_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/2020 Super Baseball (U).smc" --appendconfig /tmp/retroarch.cfg

video_refresh_rate = 60
video_fullscreen_x = 640
video_fullscreen_y = 480

for example.

joolswills, 2015-08-12:
for now I would switch the default emulator to catsfs or something, and I will look into it. It's likely to be related to some upstream changes if it didn't used to crash but now does.

Avatar2200, 2015-08-12:
Thank you for your efforts. Can't tell exactly when the problem appeared, my last version was 2.3 which worked fine (on my rpi1) with all roms that now are affected by this issue.

raedeus, 2015-08-12:
Indeed, joolswills, thank you for your efforts. In the future I should reference my version to assist in troubleshooting. I know I JUST updated the script but, to help you out moving forward, where would I reference the version number when I have cloned the git to an existing raspbian installation? I've poked around the cloned repo and couldn't find anything.

Once again, thank you for your prompt input and assistance.

joolswills, 2015-08-12:
well, it will be related to libretro snes9x-next or retroarch and nothing to do with our code - The binary was last built a week or so ago, so im assuming a chance happened somewhere before that - I'll need to bisect it to test.

joolswills, 2015-08-12:
I have found a way to solve the crashes by changing compiler flags. Will have a workaround shortly.

raedeus, 2015-08-12:
Great! Thanks.

As far as recompiling goes, could you point me in the right direction for best practices in doing so? I have installed the entire system via binaries. 

FYI: in the meantime I did test out catsfc and it does run the game in question.

joolswills, 2015-08-12:
Please can you be more specific regarding recompiling ? You mean manually, or via retropie-setup (see menu 5)

joolswills, 2015-08-12:
This is fixed now - please re-install lr-snes9x-next via menu 5 (from binary). If you want to build from source (no point - but just in case) - you will need to update retropie-setup first, or else you will create a problem binary.

raedeus, 2015-08-12:
Great! That's all I needed to know. I was jumping ahead in my question about recompiling. I understand what needs to be done.

Thanks again for the fast response and fix!

Cheers!

raedeus, 2015-08-12:
Confirmed! This fixed it!

Woot.

closed 2015-08-12 by joolswills

NizZ8, 2015-08-12:
awesome, was going to post about this but it's already fixed!

Avatar2200, 2015-08-12:
Works for me too now! You're the man! :)

Ekstreme, 2015-08-14:
Fix my issues. Much appreciated, thanks