RetroPie/RetroPie-Setup #949: "version info" retropie menu script

gonzo-the-great, 2015-08-13:
hello, would it be possible to add a new bash script in the retropie-menu,
that basically tells you what version of retropie your currently running? 

thank you

HerbFargus, 2015-08-13:
I'm not quite sure how that would be possible as it is in constant development, and it would change any time the script was updated. But I could be wrong...

gonzo-the-great, 2015-08-13:
or at least, somehow of showing the current version number

robertybob, 2015-08-13:
If it wasn't updated so often (thank god it is) then I'd suggest putting the version number within Emulationstation (e.g adding 'RetroPie 3' to the Simple theme main menu in the bottom right corner)

joolswills, 2015-08-13:
We could display the script version in regards to git revision, but not much else - and it would be a rough guideline only since you could have an updated script, but ancient binaries etc.

joolswills, 2016-02-17:
@HerbFargus has added this to the RetroPie-Setup menu now.

closed 2016-02-17 by joolswills

HerbFargus, 2016-02-17:
Resolved as of https://github.com/RetroPie/RetroPie-Setup/pull/1273

There isn't any clean way to address binary versions or anything like that so this will only apply to the current version of the setup script. At least gives users a a gauge of where they are at.