RetroPie/RetroPie-Setup #950: gamelists.xml for subfolders support

gonzo-the-great, 2015-08-13:
ok, i originally posted this over at the emulation station repo, but since the development is on a kinda hiatus, i thought i'd post here, to see if theres any help etc,

i've been trying for some time to get gamlists.xml to show/work in sub folders, with the aim to reduce the number of emulator menus on the main selection pages, so i can have a snes main folder then that containing the sub folders of eu/usa, japanese, translations, hacks and so on this will for me drastically reduce the sizes of my game lists and i'm hoping the will be a little bit more responsive too

joolswills, 2015-08-13:
would need emulationstation changes - unfortunately development on ES is slow due to the main dev being busy on other things. We have a fork of ES with some retropie specific controller stuff, but apart from that our fork isn't too customised. Still is worth putting it over at the issue tracker for ES though - https://github.com/Aloshi/EmulationStation - someone might implement it.

gonzo-the-great, 2015-08-17:
sadly i did post over there on august 5, but knowing support is kinda on hiatus, i thought i'd post here incase it was something that could be figured out retropie end, thank you tho

joolswills, 2015-10-15:
Copied to https://github.com/RetroPie/EmulationStation/issues/10

closed 2015-10-15 by joolswills