RetroPie/RetroPie-Setup #952: Idea - Option to compare latest versions/revisions of emulators to those installed

robertybob, 2015-08-13:
I always use the pre-compiled download from the website and have never bothered with compiling emulators to be sure I have the 'latest version'. Doing this for the 9 emulators I frequently use would be very time consuming, from what I understand.

I would be more included to 'update' the emulators, however without checking on Github to see when a commit was last made, I've no idea whether ​I do indeed have the latest version of a particular emulator. For example, some are updated more than others (weekly vs. yearly). 

Is there any way to have, within the Source Installation option within the RP setup script, to show all emulators with their last updated (commit) date in brackets.

OR

Have a separate list of emulators updated since the RetroPie release (date the pre-made download was built) or during the last 30 days.

No one wants to install emulators from source on a weekly or monthly to get the best features and compatibility , if they haven't actually been touched. Also would saves people religiously 'watching' the emulators on GitHub. 

Is this technically possible?

joolswills, 2015-08-13:
Yes, but it is a lot of work. It isn't going to happem anytime soon (or at all)

joolswills, 2015-08-13:
The commit dates would say very little anyway - you would still have to check on github etc to see what has changed etc. Just subscribe to the emulators you are interested in on github?

robertybob, 2015-08-13:
It's not so much what changes have been made, but *whether* changes have been made if what I'm getting at. Also I don't imagine absolutely everyone who uses RP has a GitHub account.

In the 'Notifications' bit of GH, I only see comments on issues and pull requests, not a list of commits, and 'Watching' thirty three different repos means it would be easy to miss if commits have been made.

robertybob, 2015-08-13:
But obviously I appreciate there are a lot of other things already in the pipeline :)

joolswills, 2015-08-13:
You can get email notifications on commits afaik. if people dont have a github account, how useful is it to know that a repository was updated - the update might have been a minor readme file change etc. you still need to see what has changed which is just as easily done from a browser etc.

joolswills, 2015-08-13:
Maybe we will add a feature to list recent changes at some point, but there are far more important things to do. Most users just want stuff working, and may not care that on the 3rd August, a change was made on fuse to "shared obj extension now matches the makefile extension "

robertybob, 2015-08-13:
I found this, but in order to subscribe to these type of notifications, admin rights are needed 
https://help.github.com/articles/receiving-email-notifications-for-pushes-to-a-repository/

I take your comments on board, I'll close this for the timebeing.

closed 2015-08-13 by robertybob

joolswills, 2015-08-13:
You can use some newsfeed software and add .atom to the commits url for github.. eg

https://github.com/libretro/fuse-libretro/commits/master.atom