RetroPie/RetroPie-Setup #953: RetroArch controller setup script not registering controller

raedeus, 2015-08-13:
Hello again,

Today I have a new and interesting issue:

After troubleshooting my Xbox360 Wireless controller setup the other day using the init script option for the xboxdrv userspace driver (https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-the-XBox360-controller), I was able to run the retroarch setup script and recognize the controller. Controller worked with all emulators tested. Hooray.

NOW: Today I attempted to run a game through the lr-pcsx-rearmed core after placing the scph1001.bin file in the ~/RetroPie/BIOS/ folder. The game ran, however, I no longer had controller input. Not sure if this is related, but it is the only thing that changed on my system since full functionality. I removed the .bin file, re-attempted without success. I tested other emulators (SNES) with no success as well. Checked xboxdrv. I had the same /dev/input/js(0-2) as I did when everything worked. Ran jstest --event /dev/input/js1 and all inputs registered. Ran another retroarch joypad config, confirmed it was using the userspace driver, however, no inputs registered and the whole script timed out on each input prompt.

Not sure what to do here. Perhaps a full re-install? Did I corrupt something when I threw the BIOS in there?

Thanks for your time and consideration.

joolswills, 2015-08-13:
I doubt anything is related to the BIOS - Must be something else. If you unplug and plug the controller back in during retroarch does it help?

It would also help knowing if you are running latest retropie image etc.

raedeus, 2015-08-13:
I believe it is the newest. I just updated the script as well.

Side note: on my previous issue report about the snes9x emulator not working (thanks for the awesome work), I had requested how to actually verify my versions, or images. It's tricky for me as I used the script on my own clean raspbian image, so I don't have an image version, but I just re-installed all binaries this week, as well as re-installing lr-snes9x binaries yesterday.

Upon further investigation, I have noticed something interesting:

When running the retroarch joypad config, the udev info it lists at the top does not match up with my controller assignments. Perhaps I don't understand the script correctly, but the controller I have registered is under /dev/input/js1 via the xboxdrv. While my controllers are populated into /dev/input/js(1-2) and when I run the jstest --event I get output under /dev/input/js1 with my controller.

Perhaps I am not following the process for configuring the controller correctly, but this worked just fine two days ago when I set up the controller. The init script is still in place, untouched. I am using an Xbox360 Wireless controller with an official Microsoft Wireless Transmitter.

Please let me know if any other details will help. I can't see how this is user error, but I have been the culprit of that too many times to rule it out. :)

Thanks.

joolswills, 2015-08-13:
the best way to configure controllers is via emulationstation, and then retroarch will get configured automatically.

Might be worth downloading a pre-made image to test also, in case it's something specific to your config.

raedeus, 2015-08-13:
That is interesting. When I first ran ES, I went through the config, but nothing worked in the emulators. After research, I was lead to believe that the ES interface controller config was separate from retroarch's controller config (and subsequently all emulator controlls) and I had to run that through the retropie_setup.sh. Could you clarify your statement? I want to make sure I understand this correctly.

Indeed I though about the pre-made image, but as everything was running PERFECT yesterday, and today I'm back to square one, I am trying to understand what happened. Perhaps a full binary-reinstall?

joolswills, 2015-08-13:
when you configure your controls in emulationstation, so long as the default config is in place that has a hook, a script is triggered that generates a controller config for emulationstation, and for retroarch based on the user input.

you could try a full binary install - make sure you update retropie-setup script first.

raedeus, 2015-08-13:
Alright, I will try that. However, I will state that the process you have just described is exactly what I originally performed and it did not work. I will re-attempt and let you know the results.

Thanks for your ongoing support!

joolswills, 2015-08-13:
you will need to install the userspace driver too of course, or shoulder button input won't work right etc. You can just configure a few buttons and then launch retropie-setup to install, or key a keyboard to drop to terminal and manually run retropie-setup to install the userspace driver.

raedeus, 2015-08-13:
Unfortunately, as anticipated, the exact same result as when I first ran the setup has happened. After the binary reinstall, with the userpace driver in place and a successful ES controller configure prompt and assignment, I have no controller input within any emulation.

Do I need to ask the retroarch team as to why the hook is not passing once configured in ES? Additionally, why is the retroarch config script registering Pads 0 & 1 as /dev/input/event(4,3) when they should be /dev/input/js(1-2)? What I am trying to say is the file paths that the xboxdrv is setting up for the device is not lining up with what retroarch is registering during the configuration attempts.

When I run a jstest for /dev/input/event(3,4) there is an error. When I run a jstest for /dev/input/js1, input is registered.

Should the retroarch joypad config be reading input from /dev/input/js(0-2) as those are the locations that the xboxdrv userspace drivers are assigning the devices too?

Once again, your time, patience, and efforts are greatly appreciated.

joolswills, 2015-08-13:
no it's our code regarding the hook that configures retroarch.

Please post the contents of /opt/retropie/configs/all/retroarch.cfg and  any files /opt/retropie/configs/rertoarch-joypads/ (via pastebin.com or similar)

the retroarch config script doesn't do anything with /dev/input - that's down to the system and the userspace driver.

Also please give me the configs in ~/.emulationstation

you shouldn't need to mess with the retroarch joypad configuration tool at all. In fact that could overwrite a good config with a bad one. Before I look into anything with that I want to see what's happening with the auto configuration.

raedeus, 2015-08-13:
Here is the output of the retroarch joypad config:

RetroArch [INFO] :: [udev]: Plugged pad: Xbox Gamepad (userspace driver) #2 (0000:0000) on port #0.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event4) supports 0 force feedback effects.
RetroArch [INFO] :: [udev]: Plugged pad: Xbox Gamepad (userspace driver) (0000:0000) on port #1.
RetroArch [INFO] :: [udev]: Pad #1 (/dev/input/event3) supports 0 force feedback effects.
RetroArch [INFO] :: Found joypad driver: "udev".
Found joypad driver: udev
Using joypad: Xbox Gamepad (userspace driver) #2

Joypads tend to have stale state after opened.
Press some buttons and move some axes around to make sure joypad state is completely neutral before proceeding.
When done, press 

To my understanding, the config will not get anything from those two /dev/input/.. locations as the controllers aren't assigned to them by xboxdrv. Like I said, perhaps I am not understanding this correctly, which wouldn't surprise me.

raedeus, 2015-08-13:
Didn't see your post as I was posting the joypad stuff. I will get what you requested now...

joolswills, 2015-08-13:
if you have run the retroarch joypad config before testing the joypad in retroarch, please re-install emulationstation via menu 5 (binary), then reboot, configure the joypad, then try a retroarch game. I also want the configs after this has happened - please don't run the retroarch joypad config menu option.

raedeus, 2015-08-13:
es_input
<script src="http://pastebin.com/embed_js.php?i=SLp3TRQj"></script>

es_settings.cfg
<script src="http://pastebin.com/embed_js.php?i=t4gJ8xGq"></script>

es_temporaryinput.cfg
<script src="http://pastebin.com/embed_js.php?i=gftqEWze"></script>

raedeus, 2015-08-13:
Well, apparently I need a pastebin tutorial. I'll have what you asked for in a bit. Apologies. Noob at this.

joolswills, 2015-08-13:
paste the contents into pastebin and post the urls here. Just easeir reading than having massive logs here.

raedeus, 2015-08-13:
Indeed. Tried to embed. Noob. I'll post in a bit.

raedeus, 2015-08-13:
retroarch.cfg
http://pastebin.com/2gWjg9Sg

es_input.cfg
http://pastebin.com/Ck12JRqr

es_input.cfg.bak
http://pastebin.com/bt8iM0S8

es_setting.cfg
http://pastebin.com/Mx4V7wkt

es_temporaryinput.cfg (interesting)
http://pastebin.com/FbSdNM8x

I will try your suggested process of reinstalling ES and configuring. Will repost.

joolswills, 2015-08-13:
You didn't post any files from /opt/retropie/configs/all/retroarch-joypads/ (after doing the steps above). I need to reference that against the other configs.

raedeus, 2015-08-13:
Apologies. I have all the autoconfigs in that directory from A-Z. I would assume this is because I ran the autoconfig script. Should I clear that directory?

joolswills, 2015-08-13:
yep. the autoconfig thing should not be used and isn't needed. I might disable it actually from the menus to avoid confusion.

joolswills, 2015-08-13:
if you clear the directory you will need to reinstall emulationstation to generate a new configuration. you should then end up with a single config in there.

raedeus, 2015-08-13:
Alright. Perhaps that is where this is all coming from. Let me try that and post what I get in that directory for you.

Thanks for your help, again.

joolswills, 2015-08-13:
np

raedeus, 2015-08-13:
/opt/retropie/configs/all/retroarch-joypads/XboxGamePad(userspacedriver).cfg
http://pastebin.com/6Dz7uZbf

Still no dice in an emulator. ES responds to input just fine. No controls in emulator. Have to disconnect power each time.

joolswills, 2015-08-13:
which emulator ? Better to stick to a simple example. What happens if you unplug / plug the joypad in during emulation ? does it print anything ?

joolswills, 2015-08-13:
Are you testing with a single controller plugged in also ?

raedeus, 2015-08-13:
No emulator works. Tested with PCSX, Snes9x, and Pico. 

I am using the Xbox360 Wireless Controller with the official Microsoft Transmitter. I have removed the batter pack (only way to turn off), and activated controller both before ES, after ES, before emulator loads, and after.

My head is turning as to how this was working yesterday, and now all functionality outside ES has dropped off completely.

Everything outside of ES and the emulators is static as far as the xboxdrv is concerned. I am intrigued to see that the ES controller setup populates the retroarch-joypads directory. Yet I get NOTHING within the emulations. I am indeed testing with only one controller. The hardware and environment from working to non-working is static. The only change is I have a USB flashdrive connected with a few roms I was carrying over. I have tried all this without that in and it is still the same.

joolswills, 2015-08-13:
no flashing box at the top right of the screen then?

No text printed on screen in retroarch (Let's use lr-picodrive as a test example) when you unplug/plug the controller back in ?

Have you tried from a fresh image also ? The next thing would be to get output from retroarch when running a game.. from a terminal try

```
/opt/retropie/emulators/retroarch/bin/retroarch --verbose -L /opt/retropie/libretrocores/lr-picodrive/picodrive_libretro.so --config /opt/retropie/configs/all/retroarch.cfg "/home/pi/RetroPie/roms/megadrive/SOMEROM"
```

replacing SOMEROM with the name of a megadrive rom - first make sure it runs etc.. then you can add to the end `>~/RetroPie/roms/megadrive/log.txt` and copy the log off so I can have a look. thanks.

raedeus, 2015-08-13:
Well, I can't generate the log.txt file. Perhaps it's because I have to reboot the system after each emulator launch and the file isn't written to the "disk"? Is there any way to jump out of the megadrive emulation cleanly? I was not able to do it through ES but perhaps through the command line such as your instructions there is a way?

Let me know and I will try that.

Perhaps I should just back up the downloaded_images and gamelist folders/files and start with an image and:

1. install binaries
2. install xboxdrv
3 don't even think about touching retroarch ANYTHING
4. reboot
5. run ES
6. config controller
7. test emulation (after recovering downloaded_images and gamelist files and their respective roms)

joolswills, 2015-08-13:
What happens when you try to generate the log file ? 

instead of rebooting, ssh in and do "killall retroarch" if you are unable to get any input recognised.

You should definitely try a premade image first btw, to rule out something that I haven't thought of.

raedeus, 2015-08-14:
Thanks, joolswills. I will do both tonight and let you know the results. I have a separate card I will image with v3.0-rpi2.

raedeus, 2015-08-14:
Alright. Very interesting things.

After blowing up a freshly generated SD image (v3.0-rpi2), I installed the xboxdrv, rebooted, configured in ES, and fired up a rom in lr-picodrive. Nothing. Tried to generate the log. Nothing.

Then, in a stroke of, "meh, what if I did this", I re-installed the emulationstation binary (after an updated script), rebooted, configured in ES, and voila, it's WORKS. Tested in lr-snes9x-next and works as well. 

So, not sure what in the world happened to HOSE the other installation (though it is a fresh raspbian + git script), but the SD image seems to be clear. I will rsync the scraped data and roms from the other card over to this SD image and test, but I think everything will clear out.

So, just to be perfectly sane here: I know that the PSX bios didn't kill this, but ~/RetroPie/BIOS is the correct place to to place the bios .bin file for pcsx-rearmed, correct?

Anyway, I will test the rest of the emulators/roms and comment tomorrow if I find that controls are still wonked.

Thanks again for your unwavering patience and your ongoing support. Hopefully I can get to throwing my controller in frustration of Mega Man rather than for lack of emulation functionality!

Cheers.

joolswills, 2015-08-14:
not sure why it should have not worked first time in the image and then worked later.

bios location is correct. 

:) heh

raedeus, 2015-08-14:
Alright. Everything seems to be working just fine. Not sure why the other setup just tanked.

I may hold off on the BIOS until I run into an actual bug, unless you have heard of something. Interestingly enough, when I dropped the bios in from my fileshare, it was executeable. I don't think it should be executeable as it is a .bin file. Just thoughts.

Well, thank you again for your help. I am sorry it wasn't an issue that brought up some actual relevant change needed (like my first one).

I will say, I think it may be good practice to:

1. Blow up the official SD image
2. Update retropie-setup script
3. Re-install the ES binary

... before doing anything else. That's what cleared this all up for me. Perhaps a full binary install through the script is part of the SOP and I missed it?

Oh well.

Thanks again for your help, joolswills. Hopefully I won't have anything to report in the near future. When/if I do, I am confident we can figure it out. You guys are doing a great job and this is a great project. I am sure it's bringing hundreds and thousands of peeps some good 'ole throwback joy, myself included.

Thanks for your time and efforts.

Cheers.

joolswills, 2015-08-14:
There should be no need to reinstall ES binary if you have the most recent image though. There are no changes at all since the 3.0 final and the ES module/scripts.

Thanks for the feedback :) Glad you have it all sorted.

raedeus, 2015-08-14:
This perplexes me as well but nothing worked until I re-installed the ES binary. 

Also, I didn't see a /opt/retropie/configs/all/retroarch-joypads/ directory or .cfg file until the re-install.

Could be wrong, but, those were things I noticed.

Works now. I'm happy until I find another challenge.

Cheers.

joolswills, 2015-08-14:
the directory is definitely included in the latest 3.0 image. the .cfg file is generated after setting up controls in emulationstation. the folder should be empty before that.

closed 2015-09-25 by joolswills

JDias, 2016-05-01:
Hi,
i follow this thread and i think i have some issue of this kind.
i updated the script and did a binary install.
i got some lines that got my attention like linking to stuff like
"â/opt/retopie" or "somescript.shâ" is that correct or am i in a locale mess?

the problem seems to be that the config file is generated for emulstation is not beeing generated for retroarch. (i think it was before i did the update and the install)

i will try my luck reinstalling emulstation, but maybe this info could be handy.

thanks in advance,

overling, 2016-07-21:
I have this issue but not going to update from 3.7  until the kernel issue on the newer builds is fixed.