RetroPie/RetroPie-Setup #956: Adding Bluetooth Device Functionality

Death259, 2015-08-14:
There's currently not an easy way that I've been able to find in the RetroPie-Setup for adding/removing/listing Bluetooth devices. I've written code for my own project that does just this. I will reformat the code and make changes to have it hopefully meet the standards for this project.

Death259, 2015-08-14:
Here is the first revision, @joolswills let me know what glaring issues you see. I haven't tested the code yet since I made the change to the retorpie coding format, so it might not work. I'll test it once the code gets a little more refined.

    #!/usr/bin/env bash

    # This file is part of RetroPie.
    # 
    # (c) Copyright 2012-2015  Florian Müller (contact@petrockblock.com)
    # 
    # See the LICENSE.md file at the top-level directory of this distribution and 
    # at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
    #

    rp_module_id="bluetooth"
    rp_module_desc="Configure Bluetooth Devices"
    rp_module_menus="3+"
    rp_module_flags="nobin"

    function depends_bluetooth() { 
    	getDepends bluez-utils bluez-compat bluez-hcidump blueman bluetooth
    }

    function remove_bluetooth_device() {
        local bluetoothDeviceList=$(bluez-test-device list)
    	if [ "$bluetoothDeviceList" == "" ] ; then
    		printMsgs "dialog" "There are no devices to remove."
    	else
    		local result_file=$(mktemp)
    		trap "rm $result_file" EXIT
    		readarray devs < <(bluez-test-device list | awk '{print NR; print $0}')
    		dialog --menu "Select device" 20 80 15 "${devs[@]}" 2> $result_file
    		exit_status=$?
    		deviceAcutallySelected=true
    		case $exit_status in
    			$DIALOG_CANCEL)
    			  deviceAcutallySelected=false
    			  ;;
    			$DIALOG_ESC)
    			  deviceAcutallySelected=false
    			  ;;
    		esac
    		if [ $deviceAcutallySelected == true ] ; then
    			arrayResult=$(<$result_file)
    			answer=${devs[$arrayResult+($arrayResult -1)]}
    			bluetoothMacAddress=($answer)
    			
    			if [ "$bluetoothMacAddress" != "" ] ; then
    				removeBluetoothDevice=$(bluez-test-device remove $bluetoothMacAddress)
    				if [ "$removeBluetoothDevice" == "" ] ; then
    					printMsgs "dialog" "Bluetooth device has been removed."
    				else
    					printMsgs "dialog" "An error occured removing the bluetooth device."
    				fi
    			fi
    		fi
    	fi
    }

    function list_bluetooth_devices() {	
        local registeredDevices="There are no registered devices"
        if [ "$(bluez-test-device list)" != "" ] ; then
            registeredDevices=$(bluez-test-device list)
        fi
        local activeConnections="There are no active connections"
        if [ "$(hcitool con)" != "Connections:" ] ; then
            activeConnections=$(hcitool con | sed -e 1d)
        fi
	
        printMsgs "dialog" "Registered Devices:\n\n$registeredDevices\n\n\nActive Connections:\n\n$activeConnections"
    }

    function connect_bluetooth_device() {
        bluetoothDeviceList=$(hcitool scan --flush | sed -e 1d)
        if [ "$bluetoothDeviceList" == "" ] ; then
            printMsgs "dialog" "No devices were found. Ensure device is on and try again."
        else
            result_file=$(mktemp)
            trap "rm $result_file" EXIT
            readarray devs < <(hcitool scan | tail -n +2 | awk '{print NR; print $0}')
            dialog --clear --menu "Select device" 20 80 15 "${devs[@]}" 2> $result_file
            exit_status=$?
            deviceAcutallySelected=true
            case $exit_status in
                $DIALOG_CANCEL)
                    deviceAcutallySelected=false
                    ;;
                $DIALOG_ESC)
                    deviceAcutallySelected=false
                    ;;
            esac
            if [ $deviceAcutallySelected == true ] ; then
                arrayResult=$(<$result_file)
                answer=${devs[$arrayResult+($arrayResult -1)]}
			
                bluetoothMacAddress=($answer)
			
                exec 5>&1
                step1=$(bluez-simple-agent hci0 "$bluetoothMacAddress" >&5)
                step2=$(bluez-test-device trusted "$bluetoothMacAddress" yes >&5)
                step3=$(bluez-test-input connect "$bluetoothMacAddress" >&5)
			
                step1Successful=true
                if [ "$step1" != "" ] && [ "$step1" != "Creating device filed: org.bluez.Error.AlreadyExists: Already Exists" ]; then
                    $step1Successful=false;
                fi
			
                step2Successful=true
                if [ "$step2" != "" ]; then
                    $step2Successful=false;
                fi
			
                step3Successful=true
                if [ "$step3" != "" ]; then
                    $step3Successful=false;
                fi
			
                if [ $step1Successful ] && [ $step2Successful ] && [ $step3Successful ] ; then
                    printMsgs "dialog" "Bluetooth device has been connected"
                else
                    printMsgs "dialog" "An error occurred connecting to the bluetooth device."
                fi
            fi
        fi
    }

joolswills, 2015-08-14:
Please see 

https://github.com/RetroPie/RetroPie-Setup/wiki/Shell-Style-Guide

which works as a guide - and checking existing code in other modules too (recent ones like confedt.sh) . thanks.

joolswills, 2015-08-14:
Sorry, I pasted that before I noticed -  not checked your code yet.

joolswills, 2015-08-14:
Few comments:

Please use [[ ]] (we are specifically using bash so good to use builtins)

Please use blah_blah and not camelcase for vars.

The functions need to be standalone functions ending with the name of the module, so they can be called separately. 

Please see example uses of dialog in other modules  to avoid using temporary files etc.

joolswills, 2015-08-14:
I think the logic for the step1/step2 step3 could be changed without multiple variables. nest it, and so on.

Death259, 2015-08-14:
Thanks for the feedback, I'll look at the guide and change the items you mentioned.

Death259, 2015-08-14:
@joolswills How does this look? I've tested each method individually and they appear to be working correctly. All variables should now contain underscores instead of camel casing and all functions should end with the module name. I've also changed all single square brackets to doubles and reworked the logic for step1/step2/step3.

     #!/usr/bin/env bash

     # This file is part of RetroPie.
     # 
     # (c) Copyright 2012-2015  Florian Müller (contact@petrockblock.com)
     # 
     # See the LICENSE.md file at the top-level directory of this distribution and 
     # at https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/master/LICENSE.md
     #

     rp_module_id="bluetooth"
     rp_module_desc="Configure Bluetooth Devices"
     rp_module_menus="3+"
     rp_module_flags="nobin"

     function depends_bluetooth() { 
          getDepends bluez-utils bluez-compat bluez-hcidump blueman bluetooth
     }

     function list_available_devices_bluetooth() {	
          local line 
          local mac_address 
          local device_name 
          while read line; do 
               mac_address=$(echo $line | cut -d, -f2)
               device_name=$(echo $line | cut -d, -f3)
               echo -e "$mac_address\n$device_name" 
          done < <(hcitool scan --flush | tail -n +2 | sed 's/\t/,/g') 
     }

     function list_registered_devices_bluetooth() {	
          local line 
          local mac_address 
          local device_name 
          while read line; do 
               mac_address=$(echo $line | sed 's/ /,/g' | cut -d, -f1)
               device_name=$(echo $line | sed -e 's/'"$mac_address"' //g')
               echo -e "$mac_address\n$device_name" 
          done < <(bluez-test-device list) 
     }

     function display_active_and_registered_devices_bluetooth() {	
          local registered_devices="There are no registered devices"
          [[ "$(bluez-test-device list)" != "" ]] && registered_devices=$(bluez-test-device list)
          local active_connections="There are no active connections"
          [[ "$(hcitool con)" != "Connections:" ]] && active_connections=$(hcitool con | sed -e 1d)
	
          printMsgs "dialog" "Registered Devices:\n\n$registered_devices\n\n\nActive Connections:\n\n$active_connections"
     }

    function remove_device_bluetooth() {	
        local mac_addresses=() 
        local mac_address 
        local device_names=() 
        local device_name
        local options=()
        while read mac_address; read device_name; do 
            mac_addresses+=("$mac_address") 
            device_names+=("$device_name") 
            options+=("$mac_address" "$device_name")
        done < <(list_registered_devices_bluetooth)
	
        if [[ ${#mac_addresses[@]} -eq 0 ]] ; then
            printMsgs "dialog" "There are no devices to remove."
        else		
            local cmd=(dialog --backtitle "$__backtitle" --menu "Please choose the bluetooth device you would like to remove" 22 76 16) 
            choice=$("${cmd[@]}" "${options[@]}" 2>&1 >/dev/tty) 
            [[ -z "$choice" ]] && return
		
            remove_bluetooth_device=$(bluez-test-device remove $choice)
            if [[ -z "$remove_bluetooth_device" ]] ; then
                printMsgs "dialog" "Device removed"
            else
                printMsgs "dialog" "An error occurred removing the bluetooth device. Please ensure you typed the mac address correctly"
            fi
        fi
    }

    function connect_device_bluetooth() {
        local mac_addresses=() 
        local mac_address 
        local device_names=() 
        local device_name
        local options=()
        while read mac_address; read device_name; do 
            mac_addresses+=("$mac_address") 
            device_names+=("$device_name") 
            options+=("$mac_address" "$device_name")
        done < <(list_available_devices_bluetooth)
	
        if [[ ${#mac_addresses[@]} -eq 0 ]] ; then
            printMsgs "dialog" "No devices were found. Ensure device is on and try again"
        else		
            local cmd=(dialog --backtitle "$__backtitle" --menu "Please choose the bluetooth device you would like to connect to" 22 76 16) 
            choice=$("${cmd[@]}" "${options[@]}" 2>&1 >/dev/tty) 
            [[ -z "$choice" ]] && return
		
            exec 5>&1
            local step_successful
            step_successful=$(bluez-simple-agent hci0 "$choice" >&5)
            if [[ -z "$step_successful" ]] && [[ "$step_successful" != "Creating device filed: org.bluez.Error.AlreadyExists: Already Exists" ]] ; then
                step_successful=$(bluez-test-device trusted "$choice" yes >&5)
                if [[ -z "$step_successful" ]] ; then
                    step_successful=$(bluez-test-input connect "$choice" >&5)
                    if [[ -z "$step_successful" ]]; then
                        printMsgs "dialog" "Bluetooth device has been connected"
                    fi
                fi		
            fi
		
            if ! [[ -z "$step_successful" ]] ; then
                printMsgs "dialog" "An error occurred connecting to the bluetooth device"
            fi
        fi
    }

Death259, 2015-08-15:
@joolswills I went ahead and created a pull request for this. If there's anything else you need changed, let me know.

joolswills, 2015-08-16:
merged. thanks.

closed 2015-08-16 by joolswills