RetroPie/RetroPie-Setup #957: GPSP not remembering edits

HerbFargus, 2015-08-14:
On GPSP when you go into the F10 menu and change your controls it seems that it won't ever remember edits. I was under the impression that it was supposed to save the configs here: /opt/retropie/emulators/gpsp/gpsp.cfg but it doesn't save the changes and reverts to the same controls before the change. am I missing something?

joolswills, 2015-08-14:
I symlinked the file to /opt/retropie/configs/gba/gpsp.cfg - perhaps it doesn't like that or something. Ill check when I get a free moment.

dlsquared, 2015-08-14:
Thanks jool!

demirk4n, 2015-09-25:
@HerbFargus most probably you are unable to save state as well.
I think there's an issue with the permissions because if I sudo ./gpsp directly both functions (save config & state) worked. As a workaround changing the ownership of "gpsp" folder to "pi" user worked for me. 
sudo chown -R pi:pi  /opt/retropie/emulators/gpsp/

HerbFargus, 2015-09-28:
So I've been looking into it a bit more and I've found that selecting exit from the F10 menu saves the configs whereas using escape to exit does not:

If anyone can verify this as well I would be interested in hearing about it.

Also if we wish for save states to work this commit may need to be reverted: https://github.com/RetroPie/RetroPie-Setup/commit/9de66f354eabc9bbe25d35463f40eac6d60855d1
and it also may be worth initially creating a /opt/retropie/emulators/gpsp/romdir.txt with `/home/pi/RetroPie/roms/gba`in it so that the save states are in the same folder as the roms

joolswills, 2015-09-28:
thanks. will sort that (adding the romdir.txt) - feel free to submit pull request though if you want :)

HerbFargus, 2015-09-28:
Ill test it a bit more just to make sure it actually works and I can submit a PR once I get it sorted. I should hopefully get it sorted this evening.

HerbFargus, 2015-10-03:
as of pr https://github.com/RetroPie/RetroPie-Setup/pull/1025 this should be resolved for the time being . not ideal but it would otherwise require upstream recoding. considering this closed.

closed 2015-10-03 by HerbFargus