RetroPie/RetroPie-Setup #960: [RP3.0] lr-imame4all does not write out cfg/nvram to disk

zerojay, 2015-08-15:
The libretro version of imame4all appears to not write out save games or any data whatsoever after it is run, rendering games that save user statistics (NBA Jam, 2 on 2 NHL Open Ice, etc...) useless to play.

As it is supposed to work, the MAME core should be dumping a cfg file (control configurations) and an nvram file (save data) for these games, however these never get created. Each time the game is launched, you get the legal disclaimer which is only supposed to show when there is no cfg file (the game has never been run before) and you'll see that the nvram is always reset.

With Retroarch, the mame cores are supposed to write to the [system directory]\mame\cfg and [system directory]\mame\nvram directories when the RA is closed down. I attempted to load up a cfg and nvram made from the advmame cores and it was a no go. There were also posts about how the cfg and nvram directories wouldn't be created if the system directory didn't end with a /, which I confirmed on PC (and is apparently intentional), however adding the / to the end of the system directory in /opt/retropie/configs/all/retroarch.cfg didn't appear to help anything. I also made sure to chown the directories I created to be owned by the pi user/group but that didn't help anything either.

joolswills, 2015-08-15:
If this is a bug in the libretro core it would be best to report it to libretro mame4all https://github.com/libretro/imame4all-libretro

joolswills, 2015-08-28:
Did you report this upstream as requested ?

joolswills, 2015-09-25:
Closing due to no user response.

closed 2015-09-25 by joolswills