RetroPie/RetroPie-Setup #961: [RP3.0] lr-imame4all has no key assigned for test/service buttons

zerojay, 2015-08-15:
Using the lr-imame4all core with Retroarch, players can very quickly realize that there is no ability to enter the test and/or service areas of games to make adjustments to the game. This especially affects the Midway games as by default, the game volume for their games are set extremely low - far too low for the built-in +12.0 dB volume adjustment in MAME to make the games remotely loud enough.

This bug also ties into https://github.com/RetroPie/RetroPie-Setup/issues/960, as even if we would be able to get in to the test menus to make the volume change, without 960 fixed, the volume would be reset each time the game is run.

zerojay, 2015-08-15:
Usually the key assigned for test is F2 in MAME.

joolswills, 2015-08-15:
I believe this should be reported upstream too, unless it's a bug with our configuration, but I suspect it's a limitation of the libretro core.

closed 2015-09-25 by joolswills