RetroPie/RetroPie-Setup #962: [RP3.0] lr-handy segmentation faults on launch every time

zerojay, 2015-08-15:
Running the lr-handy core with any verified good Lynx dump (No-Intro) results in the lr-handy core outputting the following:

Invalid cart.
Invalid ROM.

And then segfaults. Does not appear to make a difference if the ROMs are zipped or unzipped.

HerbFargus, 2015-08-15:
I just added a fresh image and I've had no problems. you may have to add lynxboot.img to the bios folder.

joolswills, 2015-08-15:
Works for me too.

zerojay, 2015-08-16:
I already have lynxboot.img in the BIOS folder (which is not mentioned in the Lynx setup page on the Wiki, by the way), also taken from the No-Intros rom set and it's still segfaulting on launch.

zerojay, 2015-08-16:
This appears to be an upstream bug with the original Handy core. If your Lynx ROMs do not contain a header - as No-Intros do not - Handy crashes and burns.

As there is no work being done on Handy anymore, it may be a good idea to mention on the Lynx wiki page that the core will only work with ROMs that have a header.

HerbFargus, 2015-08-16:
Here's the current repo.

https://github.com/libretro/libretro-handy

I'll make a note of it in the wiki.

closed 2015-08-25 by joolswills