RetroPie/RetroPie-Setup #965: Default SNES emulator: lr-snes9x-next issues with roms?

biscuits99, 2015-08-15:
I'm finding a lot of my snes roms wont load with lr-snes9x-next, but are fine with the previous default of lr-pocketsnes.

Others seem to have the same issue:
https://www.reddit.com/r/RetroPie/comments/3h3utz/certain_snes_roms_crashing_on_launch_retropie_30/

Here is the runcommand erorr log when running Mario Kart with lr-snes9x-next.

    PPU.RenderSub = 0
    PPU.FullClipping = 1
    Settings.Transparency = 1
    Settings.SpeedhackGameID = 0
    PPU.SFXSpeedupHack = 0
    coldata_update_screen = 1
    /opt/retropie/supplementary/runcommand/runcommand.sh: line 696:  2819 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x-next/snes9x_next_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/Super Mario Kart (U) [!].smc" --appendconfig /tmp/retroarch.cfg

This rom runs fine in lr-snes9x-next though  "Super Mario World (U) [!].smc"

Either I'm doing something wrong, or pocketsnes may be a better default.

HerbFargus, 2015-08-15:
youve tried this? https://github.com/RetroPie/RetroPie-Setup/issues/946

joolswills, 2015-08-15:
Yeh you need to update the binary.

biscuits99, 2015-08-15:
Ah - fixed! Sorry about the duplicate.

closed 2015-08-15 by biscuits99