RetroPie/RetroPie-Setup #966: "No Gamepads Detcted" after reboot

okaysolutions, 2015-08-17:
So this is really stange issue I can't figure out. 

I have a Raspberry Pi 2 running raspbian with a Keyboard and Xbox 360 controller plugged in. 

I ran the newest retropie binary setup and ran "emulationstation" afterwards and everything worked as expected. However, after I reboot and run "emulationstation" I am brought to the controller configuration menu and I get the issue "No gamepads detected"

It looks like everytime I run the binary setup it works (i can set up a controller and play games) until I reboot. 

I'm not sure what other information is needed, let me know!

joolswills, 2015-08-17:
How is the 360 controller connected? Are you using the userspace driver (Under retropie-setup supplementary menu).

joolswills, 2015-08-17:
Are you using the retropie 3.0 image (Or a recent install via retropie-setup ?)

okaysolutions, 2015-08-17:
Thanks for the quick reply :)

The 360 controller is connected via the userpace driver. I think it's also important to note that after I reboot and get the "No gamepads detected" the keyboard is not recognized either. The UI does not respond to f4. 

It was installed using the retropie-setup on raspbian.

joolswills, 2015-08-17:
Are you using a powered hub ? What version of the raspberry pi ? Do you have any flashing block at the top right of the screen ? I suspect a power issue is the problem, or some faulty USB device.

okaysolutions, 2015-08-17:
I am using the iPad (5.1v, 2.1a) charger.
Raspberry Pi 2
Yes there is a flashing block at the top right of the screen.

It's strange to me that its a power issue being that it works fine after the initial install but if you think that that could be the issue I'll try out a powered USB hub.

joolswills, 2015-08-17:
Maybe it was worth mentioning the flashing block in the original post (As it is not expected etc) - could have saved some time.

you have a power supply issue or cable - it could be that the power supply is ok, but the usb cable is not sufficient. 

https://www.raspberrypi.org/forums/viewtopic.php?p=589877#p589877

okaysolutions, 2015-08-17:
Alright, now I'm using a usb hub to power the pi and the controller. There is no more rainbow box but the issue still persists.

joolswills, 2015-08-17:
have you tested keeping either the joypad or keyboard disconnected. Does it happen with just the keyboard connected. To be honest, this doesn't look like a retropie issue, but perhaps some hardware issue with your usb devices.

okaysolutions, 2015-08-17:
The keyboard and controller work outside of emulationstation which leads me to believe it's a software problem, not a hardware one.

joolswills, 2015-08-17:
Please try my first two suggestions anyway.

okaysolutions, 2015-08-17:
It doesn't work with either of them plugged in individually.

joolswills, 2015-08-17:
eg. does f4 work on the keyboard to quit ES, when just the keyboard is used. Or does the controller still misbehave when there is no keyboard detected (in ES). does it make a difference if you disable autostarting of emulationstation, (via supplementary menu), and then loading it manually after waiting a while?

joolswills, 2015-08-17:
Please also try a prebuilt retropie image to see if you have the same problem, in case it's something with your raspbian install.

okaysolutions, 2015-08-17:
I'm currently not autostarting emulationstation. I was just starting it manually. And the pressing f4 on the keyboard does not do anything. 

Trying the prebuilt image was my next step, it just required buying more hardware so I was hoping to avoid that. I'll let you know.

joolswills, 2015-08-17:
You are starting emulationstation manually

you could backup your SD card re-use it. But SD cards are cheap anyway, so better to get some spares.

you could post me the output of "dmesg" (dmesg >/some/path/dmesg.log after loading emulationstation too (via pastebin.com or similar). Also posting ~/.emulationstation/es_log.txt might help.

okaysolutions, 2015-08-17:
http://pastebin.com/0PUvKYz5

joolswills, 2015-08-17:
What version of raspbian is this - or have you manually upgraded the kernel to a bleeding edge kernel ? Please test with the latest packaged kernel and not the 4.x

okaysolutions, 2015-08-17:
I don't remember what version it is. I haven't modified the kernel at all. The one installed now should be the ones i've gotten from regular updates & upgrades

➜  ~  uname -a
Linux buoy 4.0.9-v7+ #807 SMP PREEMPT Fri Jul 24 15:21:02 BST 2015 armv7l GNU/Linux

joolswills, 2015-08-17:
You shouldn't get that kernel via standard updates with apt-get etc. Did you use rpi-update?

I would advise against using rpi-update unless there is something specifically you need. Kernels are packaged up now and again when they are deemed ready.

https://www.raspberrypi.org/forums/viewtopic.php?p=441963#p441963

andrewsil1, 2015-08-17:
Speaking from experience, the iPad charger is definitely not sufficient to run a Raspberry Pi 2 and a few USB peripherals consistently at load without showing the brownout indicator that you’re seeing.  A powered hub will likely help the situation by offloading the peripherals’ power consumption, but I believe it’s still possible that with all 4 cores running full tilt, the iPad charger may still wind up sagging.  At least you get the visual indication of when there’s a problem, so you can try the powered hub first and take it a step at a time.
_____________________________
Andy Silverman

okaysolutions, 2015-08-17:
@andrewsil1 I've tried powering it though the hub as well with the only peripheral being the xbox 360 controller with no luck. It's just strange to me that the controller and the keyboard work outside of emulation station and work right after install of this is strictly a power issue.

joolswills, 2015-08-17:
Well, it still could be related to power - the CPU is more loaded when ES is running which could cause a problem if there was not enough power to the USB interfaces etc.

I would like to see how it behaves also with a 3.18.x kernel. The one you are running is neither the latest bleeding edge (via rpi-update), nor the packages 3.18 kernel, so it may be connected.

okaysolutions, 2015-08-17:
I could see it being a kernel issue. I've been meaning to reformat anyways. I'll probably give that a shot.

okaysolutions, 2015-08-25:
I tried the exact same hardware configuration with the retropie standalone and it works fine.

joolswills, 2015-08-25:
Must have been the kernel then - or some other customisation. I shall close then. BTW there is a new upstream kernel out now. a 4.1 - worth doing apt-get update; apt-get dist-upgrade and make sure it still works.

okaysolutions, 2015-08-25:
Sure thing. I'll let you know if I figure out what part of my configuration is messing it up

closed 2015-08-25 by okaysolutions

okaysolutions, 2015-08-26:
Just installed a 3.18 kernel on the original SD card and have the same issue. This is very peculiar. For now I can just deal with having a seperate SD card but this is really strange

joolswills, 2015-08-26:
must be something else on the original sd card - some config.txt option or something else.

tmar89, 2015-10-28:
So I am having a very similar issue. When I update the Retropie scripts and do a binary update and reboot, ES doesn't detect my controllers anymore but the .cfg files still exist in /opt/retropie/configs/all/retroarch-joypads. And when I map a controller I already had in the previous version of retropie, it makes a .bak file of the previous configuration file. How come the original config files in that folder aren't recognized when the controller is plugged in after an update? I've tried this with all my USB controllers and the keyboard also.

joolswills, 2015-10-28:
ES currently does wipe its own controller config during an install, so you have to reconfigure. It wont get rid of the retroarch-joypads config though - emulationstation doesn't use that - that's for retroarch emulators.

I will probably change the ES install script to preserve existing configs, but in the meantime, just reconfigure etc.

Your issue is unconnected to this closed ticket though.

tmar89, 2015-10-28:
Ok thanks, joolswills. That makes sense. I was searching around endlessly and this was the closest topic I could find to what I was experiencing. I think it would be a great idea to maintain existing configs with an install. Sorry for disturbing a closed ticket but thanks for replying :+1: