RetroPie/RetroPie-Setup #967: Netplay

expandables2, 2015-08-17:
Hi good day, I see netplay is available but has anyone done a netplay? Also how does netplay work. Thanks!

joolswills, 2015-08-17:
This would be best posted on the forum rather than this issue tracker.

joolswills, 2015-08-17:
Please post it at the RetroPie forum - http://blog.petrockblock.com/retropie/

closed 2015-08-17 by joolswills

HerbFargus, 2015-08-17:
I've done it over my local network but I haven't been able to use it over my external network as my ISP is restrictive. You just put the same netplay configs for both except one as a host  and one is a client, then run one with netplay enabled (using the run command menu) and it will wait until the other runs the same game with netplay enabled. It has to be the exact same ROM on the exact same emulator. I'll probably create a wiki page on it once I've got time. In the mean time though github is used for bug reports, the retropie forum is a better place for your question.

http://blog.petrockblock.com/forums/forum/retropie-project-forum/