RetroPie/RetroPie-Setup #969: Video Splashscreen bug

HerbFargus, 2015-08-19:
I know splashscreens are your favourite things to deal with... but a user ran into a big issue with video splash screens.

In the past we could remove the `&` after do start in order to allow the video to play all the way through without emulationstation cutting into it, but when editing asplashscreen as root it gives the error 

`touch: cannot touch '/tmp/writable': Read-only file system` 

and no longer boots at all...

joolswills, 2015-08-19:
It would have been useful to mention this "feature" when all the work was being done :)

I'll look into it.  I wouldn't call it a big issue though :)

joolswills, 2015-08-19:
I can see why that won't work now - but then it is a hack to hold up the booting like that - it would need to be implemented differently. Might just be easier to add a sleep to /etc/profile.d/10-emulationstation or so to pause the loading of emulationstation rather than hacking the init.d

joolswills, 2015-08-19:
this may be better - to add to the top of the same file

```
while pgrep omxplayer &>/dev/null; do sleep 1; done
```

better than editing the init.d

HerbFargus, 2015-08-19:
I know, I should have tested it more thoroughly- I had forgotten about that little tidbit. you're brilliant mate:

`/etc/profile,d/10-emulationstation.sh`

add the line so it looks like this:
```
while pgrep omxplayer &>/dev/null; do sleep 1; done
[ "`tty`" = "/dev/tty1" ] && emulationstation
```

works perfectly.

closed 2015-08-19 by HerbFargus

joolswills, 2015-08-19:
I don't mind making that a default if you think it is the most common usecase (that it waits for the video). What do you think ?

HerbFargus, 2015-08-19:
I think most video boot sequences will be longer than 4 seconds- so I think its wise. if people do want it cut into by emulationstation they could remove the line.