RetroPie/RetroPie-Setup #972: Links/aliases to important directories

ProxyCell, 2015-08-19:
Would it be possible to have an optional setup script that adds in some links/aliases/shortcuts to the important directories directly in the $homedir

joolswills, 2015-08-19:
I don't feel this is needed as a built in thing. What is your use case ?

ProxyCell, 2015-08-19:
well, just faster/easier to traverse to some harder to reach folders when accessing the Pi over SFTP
it would let me drag and drop some files

joolswills, 2015-08-19:
If you need that I would just make the symlinks yourself (if you have tested that works). I don't want to start making symlinks in users folders by default for the sake of making drag and drop easier on a specific sftp client.

ProxyCell, 2015-08-19:
good point, other users may want their homedir cleaer

closed 2015-08-19 by ProxyCell