RetroPie/RetroPie-Setup #973: [RP3.0] - ScummVM "Error Running Game"

NizZ8, 2015-08-19:
I'm getting a "Error running Game: Game data not found" error when i try to launch a game.  If i launch it directly in ScummVM it will start correctly, just not through the ES UI.  I tried deleting the .svm's and re-creating them but that didn't work either.

NizZ8, 2015-08-19:
Disregard--Issue was fixed when i did a reboot after removing the games from the scummvm app and then re-adding them.

closed 2015-08-19 by NizZ8

HerbFargus, 2015-08-19:
I've had similar issues- I wonder why it is that it doesn't remember the settings.

HerbFargus, 2015-10-09:
@joolswills 

worth reopening to just give a second look to make sure that this isnt a bug as I've yet to get it working on mine.

joolswills, 2015-10-09:
thanks. will check.

HerbFargus, 2015-10-10:
tested it again on the jessie build following the suggestion to reboot above and it works with no problems. no idea why it wasn't working on my other build, little weird but whatever. Glad it works now.