RetroPie/RetroPie-Setup #976: PS3 Controller not being detected vis usb in RetroPie 3.0

tigarcia, 2015-08-21:
lsusb output:
Bus 001 Device 007: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller

jstest /dev/input/js0:
Does not seem to be updating after intial dump of buttons.

Tried Installing/Pair PS3 Controller. It asked for the bluetooth dongle which I do not have.  I had to cancel.

tigarcia, 2015-08-22:
I should also mention I have a raspberry pie 2

joolswills, 2015-08-22:
The pairing is for connecting via bluetooth.

The full output form `dmesg` may help. What is your pi power source ?

joolswills, 2015-08-22:
do you have a rainbow block at the top right of the screen ?

tigarcia, 2015-08-22:
The pi power source is 2.5amp power supply from the canakit.

Below is the dmesg output:

```
[    2.299135] usb usb1: SerialNumber: bcm2708_usb
[    2.309847] hub 1-0:1.0: USB hub found
[    2.318926] hub 1-0:1.0: 1 port detected
[    2.328538] dwc_otg: FIQ enabled
[    2.328552] dwc_otg: NAK holdoff enabled
[    2.328563] dwc_otg: FIQ split-transaction FSM enabled
[    2.328604] Module dwc_common_port init
[    2.328997] usbcore: registered new interface driver usb-storage
[    2.340556] mousedev: PS/2 mouse device common for all mice
[    2.351924] bcm2835-cpufreq: min=600000 max=900000
[    2.362296] sdhci: Secure Digital Host Controller Interface driver
[    2.373618] sdhci: Copyright(c) Pierre Ossman
[    2.383317] DMA channels allocated for the MMC driver
[    2.427888] Load BCM2835 MMC driver
[    2.443901] sdhci-pltfm: SDHCI platform and OF driver helper
[    2.455350] ledtrig-cpu: registered to indicate activity on CPUs
[    2.466760] hidraw: raw HID events driver (C) Jiri Kosina
[    2.477626] usbcore: registered new interface driver usbhid
[    2.489423] usbhid: USB HID core driver
[    2.500764] TCP: cubic registered
[    2.511262] Initializing XFRM netlink socket
[    2.520670] NET: Registered protocol family 17
[    2.530388] Key type dns_resolver registered
[    2.538017] Indeed it is in host mode hprt0 = 00021501
[    2.550446] Registering SWP/SWPB emulation handler
[    2.561418] registered taskstats version 1
[    2.570891] vc-sm: Videocore shared memory driver
[    2.580722] [vc_sm_connected_init]: start
[    2.590512] [vc_sm_connected_init]: end - returning 0
[    2.602009] Waiting for root device /dev/mmcblk0p2...
[    2.617688] mmc0: host does not support reading read-only switch, assuming write-enable
[    2.634472] mmc0: new high speed SDHC card at address e624
[    2.645891] mmcblk0: mmc0:e624 SL08G 7.40 GiB 
[    2.657157]  mmcblk0: p1 p2
[    2.717945] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.729897] Indeed it is in host mode hprt0 = 00001101
[    2.743363] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    2.756889] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    2.781311] devtmpfs: mounted
[    2.790225] Freeing unused kernel memory: 384K (80754000 - 807b4000)
[    2.938331] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    2.950588] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.964098] hub 1-1:1.0: USB hub found
[    2.973544] hub 1-1:1.0: 5 ports detected
[    3.258138] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.368439] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    3.381619] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.398406] smsc95xx v1.0.4
[    3.472276] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:87:ea:39
[    3.568137] usb 1-1.2: new full-speed USB device number 4 using dwc_otg
[    3.686640] usb 1-1.2: New USB device found, idVendor=05ac, idProduct=1003
[    3.699978] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.714229] usb 1-1.2: Product: Hub in Apple Extended USB Keyboard
[    3.727096] usb 1-1.2: Manufacturer: Mitsumi Electric
[    3.739945] hub 1-1.2:1.0: USB hub found
[    3.751719] hub 1-1.2:1.0: 3 ports detected
[    3.944622] udevd[175]: starting version 175
[    4.049569] usb 1-1.2.2: new full-speed USB device number 5 using dwc_otg
[    4.249161] usb 1-1.2.2: New USB device found, idVendor=054c, idProduct=0268
[    4.266095] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.288230] usb 1-1.2.2: Product: PLAYSTATION(R)3 Controller
[    4.301221] usb 1-1.2.2: Manufacturer: Sony
[    4.314070] usb 1-1.2.2: rejected 1 configuration due to insufficient available bus power
[    4.329942] usb 1-1.2.2: no configuration chosen from 1 choice
[    4.344070] usb 1-1.2.2: No support over 100mA
[... 13 lines cut ...]
[    4.974629] Bluetooth: L2CAP socket layer initialized
[    4.986539] Bluetooth: SCO socket layer initialized
[    5.609270] random: nonblocking pool is initialized
[    6.436698] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.781509] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   11.821533] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.433246] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   15.122148] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS
[   15.619541] Bluetooth: RFCOMM TTY layer initialized
[   15.619604] Bluetooth: RFCOMM socket layer initialized
[   15.619651] Bluetooth: RFCOMM ver 1.11
[   15.635144] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.635165] Bluetooth: BNEP filters: protocol multicast
[   15.635198] Bluetooth: BNEP socket layer initialized
[  166.407560] usb 1-1.2.2: USB disconnect, device number 5
[  170.370632] usb 1-1.2: USB disconnect, device number 4
[  170.370670] usb 1-1.2.3: USB disconnect, device number 6
[  176.754696] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  176.948941] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[  176.948973] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  176.949013] usb 1-1.2: Product: PLAYSTATION(R)3 Controller
[  176.949031] usb 1-1.2: Manufacturer: Sony
[  177.118019] input: Sony PLAYSTATION(R)3 Controller as /devices/platform/bcm2708_usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0003/input/input2
[  177.121780] sony 0003:054C:0268.0003: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-bcm2708_usb-1.2/input0
```

tigarcia, 2015-08-22:
I haven't noticed a rainbow block.  Which screen would I see it on?  Do you have an image of what it would look like?

joolswills, 2015-08-22:
http://i57.tinypic.com/24fxx6t.png

You have a power issue anyway. Either need a new PSU or better USB cable. 

```
rejected 1 configuration due to insufficient available bus power
```

joolswills, 2015-08-22:
Are you using a hub ? Could be you need a configuration change for it too

https://github.com/raspberrypi/linux/issues/273

joolswills, 2015-08-22:
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=66768

tigarcia, 2015-08-22:
I think the rejected message may have been a red herring.  Sorry about that. I initially had the controller plugged into the addition usb port on my old apple keyboard.  During that same run, I took the controller out and plugged it into the usb directly.

I started up again without a keyboard and without hdmi plugged in.  My pi just has the controller plugged directly into usb, ethernet, and a 2.5amp power supply.  I ssh-ed to my pi and got the output from `dmesg`.  I didn't see anything like the rejected line.  The PS3 controller is detected and I see it listed in `lsusb`.  Here is the `dmesg` output:

```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B
[    0.000000] cma: Reserved 8 MiB at 0x2e800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 192512
[    0.000000] free_area_init_node: node 0, pgdat 8080f480, node_mem_map ae213000
[    0.000000]   Normal zone: 1504 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 192512 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (8620->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 10 pages/cpu @ae1e1000 s11456 r8192 d21312 u40960
[    0.000000] pcpu-alloc: s11456 r8192 d21312 u40960 alloc=10*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 191008
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa21041 bcm2709.serial=0x3887ea39 smsc95xx.macaddr=B8:27:EB:87:EA:39 bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 745760K/770048K available (5722K kernel code, 397K rwdata, 1748K rodata, 384K init, 763K bss, 24288K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xaf800000 - 0xff000000   (1272 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xaf000000   ( 752 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80753a48   (7471 kB)
[    0.000000]       .init : 0x80754000 - 0x807b4000   ( 384 kB)
[    0.000000]       .data : 0x807b4000 - 0x808174bc   ( 398 kB)
[    0.000000]        .bss : 0x808174bc - 0x808d6254   ( 764 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000014] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424256ns
[    0.000036] Switching to timer-based delay loop, resolution 52ns
[    0.000329] Console: colour dummy device 80x30
[    0.001768] console [tty1] enabled
[    0.001818] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001900] pid_max: default: 32768 minimum: 301
[    0.002307] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002368] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003645] Initializing cgroup subsys memory
[    0.003733] Initializing cgroup subsys devices
[    0.003787] Initializing cgroup subsys freezer
[    0.003838] Initializing cgroup subsys net_cls
[    0.003900] Initializing cgroup subsys blkio
[    0.004009] CPU: Testing write buffer coherency: ok
[    0.004126] ftrace: allocating 19614 entries in 58 pages
[    0.052604] CPU0: update cpu_capacity 1024
[    0.052680] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.052719] [bcm2709_smp_prepare_cpus] enter
[    0.052874] Setting up static identity map for 0x528478 - 0x5284ac
[    0.112514] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.112825] CPU1: Booted secondary processor
[    0.112834] [bcm2709_secondary_init] enter cpu:1
[    0.112886] CPU1: update cpu_capacity 1024
[    0.112895] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.132484] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.132739] CPU2: Booted secondary processor
[... 168 lines cut ...]
[    4.604052] input: Sony PLAYSTATION(R)3 Controller as /devices/platform/bcm2708_usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:054C:0268.0001/input/input0
[... cut ...] off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:ofAxes:  0:-32767  1:-32767  2:-32767  3:-32767  4:-32767  5:-32767  6:-32767  7:-32767  8:-32767  9:-32767 10:-32767 11:-32767 12:-32767 13:-32767 14:-32767 15:-32767 16:-32767 17:-32767 18:-32767 19:-32767 20:-32767 21:-32767 22:-32767 23:-32767 24:-32767 25:-32767 26:-32767 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off 13:off 14:off 15:off 16:off 17:off 18:off
```

joolswills, 2015-08-23:
I have no idea. I have a ps3 controller here and don't have the problem - perhaps try another controller ?

joolswills, 2015-08-23:
are you tested with a fresh 3.0 image ? Have you tried this controller on another PC to confirm it is working fine ? Not sure I can help further - perhaps there is some hardware problem - It isn't anything relating to retropie though if the controller isn't working from the low level tools

tigarcia, 2015-08-23:
I figured out the issue.  It turns out that whenever I connected my ps3 controller to the pie via usb, it would automatically turn the ps3 on and pair with the console.  Whenever the ps3 is on, the controller does not respond with the pie.  I had to unplug the ps3 while using the controller with retropie.

Is there anywhere in the wiki I should mention this?  I have been debugging for a while!

HerbFargus, 2015-08-23:
https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-PS3-controller

joolswills, 2015-08-23:
Always worth mentioning everything when posting an issue - like having your ps3 switched on / controller paired :)

but glad you have it sorted.

closed 2015-08-23 by joolswills

tigarcia, 2015-08-24:
Actually the PS3 game console was off.  That's why I didn't notice it.  Whenever the pi boots, the controller gets power and must send a wake up signal to the ps3.  I had to actually pull the power cord out of the ps3 to stop it from happening.  I'll be explicit in the wiki.

foocp, 2016-12-13:
+1 
the controller is off and you don't even notice it. thanks for the tip!