RetroPie/RetroPie-Setup #977: Can this be done? As a future update..

Haris1977, 2015-08-22:
I pulled it from within RecallBox (configure menu for p2,p3 and p4):

http://www.2shared.com/photo/UDXeEWYg/Controller.html

+1 for this. I think it needs to be implemented into future updates..

Tele42, 2015-08-22:
It's generally considered good form to type out (almost) everything so that an issue report stands on it's own without any external references, that way the issue report is still coherent if the other website removes whatever content in the future. Having said that, this issue seems to be for Emulationstation's upstream development (https://github.com/Aloshi/EmulationStation/issues).

joolswills, 2015-08-22:
They have some custom controller stuff in recallbox. We don't yet have any scripts to configure fba etc, so no point in player 2-4 controls yet, but the code iabove is specific to recallbox, and would not work without changes against our already modified es I guess. (We have some of our own custom controller stuff)

Haris1977, 2015-08-22:
As i am little baffled here..can anyone explain how to setup a second (different) controller? If i do it from within ES menu-having plugged in my 1rst controller-will i be able to configure the second one? or should i remove my first gamepad and then do it?

joolswills, 2015-08-22:
You can just go to the ES menu and configure another controller, and it will set up a retroarch configuration for it, and make it work with ES too.

joolswills, 2015-08-22:
You can leave your first controller plugged in.

closed 2015-09-28 by Haris1977