RetroPie/RetroPie-Setup #978: Add controller config picture to rom section

secsven, 2015-08-22:
hey,

this is a feature request to add the controller config overview as shown here:
https://github.com/RetroPie/RetroPie-Setup/wiki/RetroArch-Configuration

to the menu

Thanks

joolswills, 2015-08-22:
Not sure I understand your request sorry.

joolswills, 2015-08-22:
oh - you want the pictures of the controllers added to retropie-setup ? Surely it's easy enough just to reference the wiki - and easier for us to maintain? Or you mean something else ?

secsven, 2015-08-22:
yes exactly, I do not always have internet, or the wiki available when running retropie ... so I thought it might would be an enhancement to have the controller overview available via the GUI

HerbFargus, 2015-08-22:
@taalas has discussed an interactive controller config integrating the diagrams that he's created- but I don't know if that will happen anytime soon. Ideally it should be integrated with retroarch rather than retropie or emulationstation

joolswills, 2015-08-25:
I don't really want to include the images in retropie-setup - as it's just another thing to maintain then - if you don't have internet access perhaps it could be useful to print the ones you want ?

closed 2015-08-25 by joolswills