RetroPie/RetroPie-Setup #979: Wrong joypad cfg files

makmonty, 2015-08-22:
Hi there.

I have three gamepads: an 8Bitdo NES Pro, an Idroid:con and an Ipega controller.
When I use the retropie_setup.sh script and go to Setup -> Configure input devices for RetroArch, it creates the .cfg files with the wrong name. In the case of the 8Bitdo controller, the file is called 8BitdoNESPro.cfg but, when I run an emulator, it looks for the 8Bitdo_NES_Pro.cfg file. Notice the underscores. The same thing happens with the Ipega controller, its cfg file is called BluetoothController.cfg, but will only work if I rename it to Bluetooth_Controller.cfg. It does not happen with the Idroid:con as it doesn't contain spaces on its name.

I have a related issue with the EmulationStation startup configuration screen, that overwrites those files (with the right name) with the wrong information, but I guess this is not the place to explain it. Or is it?

Thank you.

joolswills, 2015-08-23:
Did you start from a retropie 3.0 image ?

I have tested and on my system the files are created with hyphens in place of spaces by our config generator (that is triggered from ES) and the retroarch joystick input option. Note it shouldn't be required to use the retroarch joystick input option as configurations should be set up from emulationstation. It is just there for those that want to manually tweak stuff.

joolswills, 2015-08-23:
actually, sorry, spaces are stripped by both, but the configs are loaded correctly from retroarch. Have you updated retroarch - if you think the wrong configs are being loaded, do you have some logging to show this ?

joolswills, 2015-08-23:
Bit confused that you are saying configs are overwritten by ES but with the right name, as you said the names are different?

joolswills, 2015-08-23:
please show me the contents of /opt/retropie/configs/all/retroarch.cfg /opt/retropie/configs/all/retroarch-joypads as well as the aforementioned logfile where you see the wrong config being loaded.

makmonty, 2015-08-23:
I didn't know there was a new version! I just updated to Retropie 3 and now it works as expected. Nice work! I was really upset with this issue as it forced me to do tricks everytime I switched on my machine. Thank you!

joolswills, 2015-08-23:
cheers.

closed 2015-08-23 by joolswills