RetroPie/RetroPie-Setup #982: Input Bluetooth Controller error

bibehck, 2015-08-27:
RetroPie 3.0
Nes30 Pro (firmware 1.63)
Raspberry Pi 2
Dongle Bluetooth 4.0 Inateck

Hi!

I followed [this tutorial](http://forum.8bitdo.com/thread-328-1-1.html) to connect my NES30 Pro controller via Bluetooth. 
All seems to work, from installing all necessary packages to trust device...

but when I typed:
```powershell
sudo bluez-test-input connect xx:xx:xx:xx:xx:xx
```

I get this error:
```powershell
Traceback (most recent call last):
  File "/usr/bin/bluez-test-input", line 40, in <module>
    input.Connect()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "Connect" with signature "" on interface "org.bluez.Input" doesn't exist
```

I look around the web, the unique answer I find is to check if in my Bluetooth's configuration the **inpu plugint** is activate. And it's ok. Any idea? :)

craigerskine, 2015-08-27:
I gave up on pairing these controllers months ago. I tried everything and kept very detailed notes but nothing allowed for consistent use. I have an NES30, SNES30, SFC30, and NES30 pro and one time back on Beta3 I got the NES30 pro to pair and re-pair consistently. I think the latest release allows for much better pairing without having to install anything extra.

There is an awesome script called [PiAssist](https://github.com/Death259/PiAssist) that can help with the pairing if you still want to go down that road.

The non-pro controllers do not allow Select + L/R for some reason which sort of makes them unusable IMO.

I've since moved on to using a DualShock 4 and I'm quite happy now. Touchpad as the hotkey modifier works pretty great.

bibehck, 2015-08-27:
Yeah, 

I already use PiAssist, very nice script. But I still get the same error (same commands) , I can't connect my controller. It's strange because first I tried to pair my pad, it's work. But since few days I get this issue and I didn't change anything. 

 I have re-flash my SD to start from scratch, and nothing change.... Very weird.

joolswills, 2015-08-27:
This is not really a retropie issue, but some problem with raspbian and your device. Without having one I can't test/come up with a fix though.

bibehck, 2015-08-27:
Yeah,  I think so. 
I thought it was more to find someone who would have been confronted with the problem here. :)

xBurningHero, 2015-09-03:
I had the same issue yesterday. I suspect this will help:

Firstly. I would create an export variable for your .bashrc with the mac address of the controller saved:
`export nespro='13:A1:FD:52:6A:61'` (or whatever your mac address is for the controller. 
You can find out with a quick:
`sudo hcitool scan`

If you have any existing device connections for the NES30 Pro you will want to remove them:
`sudo bluez-test-device remove ${nespro} `

Then:
`sudo update-rc.d -f dbus defaults` - I suspect this is the actual issue with your current configuration.
Then reboot.
Once you're back in a bash session:
`sudo hcitool dev` <= Take note of the interface name. Usually hci0
`sudo bluez-simple-agent hci0 ${nespro} `
`sudo bluez-test-device trusted ${nespro} yes ` 
`sudo bluez-test-input connect ${nespro} `

Hoping that helps. 
At this time the firmware for the nespro 30 doesn't really support the auto-connect feature. As such pairing on boot can be a hassle. There is an unfortunate feature of EmulationStation that force you to reconfigure the controller if it is not detected once ES actually starts. To bypass this I keep a usb controller plugged into the system and created a cronjob that checks if the device is paired and then connects if not. 

However, as for the RetroPie issue: It would be really great if we could bypass the automatic controller configuration if no pads are detected screen once it has been detected.

joolswills, 2015-09-03:
you can create an empty keyboard configuration for ES (or just configure a keyboard), and it will never prompt if a controller is missing.

joolswills, 2015-09-03:
eg

```
<?xml version="1.0"?>
<inputList>
  <inputAction type="onfinish">
    <command>/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh</command>
  </inputAction>
  <inputConfig type="joystick" deviceName="Logitech Logitech Cordless RumblePad 2">
    <input name="pagedown" type="button" id="7" value="1"/>
    <input name="start" type="button" id="9" value="1"/>
    <input name="pageup" type="button" id="6" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="a" type="button" id="1" value="1"/>
    <input name="b" type="button" id="2" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="select" type="button" id="8" value="1"/>
    <input name="left" type="hat" id="0" value="8"/>
  </inputConfig>
<inputConfig type="keyboard" deviceName="Keyboard">
    <input name="start" type="button" id="9" value="1"/>
</inputConfig>
</inputList>
```

joolswills, 2015-09-03:
See this thread for more info http://blog.petrockblock.com/forums/topic/bluetooth-connection-prompt

bibehck, 2015-09-03:
Thanks xBurningHero & joolswills, I will try both tricks tomorrow. It's seems nice. :)
For now, I use `hidd --connect` with an auto script (btDaemon.sh), it's works like a charm. I made a tutorial with both methods on my blog http://bibehck.fr/nes30-pro-retropie-3-0/ (in french o/) 

https://translate.google.com/translate?sl=fr&tl=en&js=y&prev=_t&hl=fr&ie=UTF-8&u=http%3A%2F%2Fbibehck.fr%2Fnes30-pro-retropie-3-0%2F&edit-text=&act=url

bibehck, 2015-09-03:
Mm... @xBurningHero I just tried your method, It don't work.

xBurningHero, 2015-09-03:
@bibehck I'm sorry to hear that. Maybe post about it on the RetroPie Forum and send me the link; I can try to take a look at the output and offer some suggestions.

@joolswills Thanks! But will that require actually having a keyboard plugged in to bypass? (Since I already have a USB player 2 remote it's now more of a non-issue, but if I have to boot without that controller or any open USB ports I'm forced to remap the Bluetooth controller once it's paired.

joolswills, 2015-09-03:
@xBurningHero no keyboard needed.

xBurningHero, 2015-09-03:
Great. Thanks @joolswills

closed 2015-09-25 by joolswills

sklivvz, 2015-12-26:
@xBurningHero  your solution doesn't work at all. Connecting two popular pads with a popular distro should not be an issue. That it is, and that there's no one that even knows what's going on (or how to debug the issue) is beyond amazing. Why is this thread closed?

joolswills, 2015-12-26:
Because the issue is resolved.

joolswills, 2015-12-26:
@sklivvz you also give no information as to what problem you are having - there are multiple issues raised in this ticket. Pairing is resolved as we have an updated bluetooth module which handles connecting to these controllers, and ES now allows a controller to be added "late" to skip the input configuration.

If you have a bug open a new ticket, but if you need support, use the retropie forum. thanks.

sklivvz, 2015-12-26:
@joolswills [done](https://github.com/RetroPie/RetroPie-Setup/issues/1153). The error looks exactly the same, but the fix does not work.

joolswills, 2015-12-26:
Also it's best not to jump to conclusions without giving people a change to reply. Your initial post above hardly helps us help you - and then you followed it up with a sarcastic tweet I see. Charming.

sklivvz, 2015-12-26:
@joolswills removed the tweet. Anything else that disturbs you?

joolswills, 2015-12-26:
Comments like that.

joolswills, 2015-12-26:
Hence, I shall clean this ticket up. No need for it.