RetroPie/RetroPie-Setup #984: Logitech f310 mapping problem

Joec85, 2015-08-27:
Hello, 

I've just set up my rpi2 to run retropie and I'm having a small problem mapping my controller. I plug it in and emulation station recognizes the f310. As I'm mapping buttons everything works fine except the right trigger button. I press this one and the system goes to skip mapping. I've tried several times and this is always the issue. Anyone have a solution?

joolswills, 2015-08-27:
Please see http://blog.petrockblock.com/forums/topic/logitech-f310-controller-cant-assign-control-to-top-right-trigger/#post-104784

adrianmoisey, 2015-09-21:
I've got a F310 and it works perfectly.
Is this the emulationstation config or the RetroArch config that you're having issues with?

joolswills, 2015-09-25:
Closing due to no user feedback (and solution is on link posted).

closed 2015-09-25 by joolswills

HerbFargus, 2015-09-29:
I can confirm the solution linked works.