RetroPie/RetroPie-Setup #986: [RPI2] switching to the latest FBA core?

nemo93, 2015-08-30:
Hi Guys,

I've been playing for a while with the latest FBA libretro core on my pi2 lately. And I have to say that this core is WAY better than the current one shipped with Retropie. Beware I'm talking about this core (https://github.com/libretro/libretro-fba) as the other one is deprecated or something. It allows this:

1. All Neo Geo sound issues are gone (KizunaE, MS3, etc). GONE!
2. Konami and Banpresto games do work flawlessly (Asterix yeah and Sailor Moon)
3. Capcom most recent games just work. I just couldn't believe it! Jojo, SF 3rd Strike, etc do work full speed with sound. Perhaps few glitches here and there with Jojo and the sound sometimes is not crackling but let's say weird (3rd selection screen for instance).
4. IGS/PGM games do work also. Amazingly. Including the fantastic Demon Front. As for the point above perhaps the sound is "weird" at random but it's all very playable and enjoyable.

Obviously this core won't work on pi1 I guess... Compilation and installation was a no brainer. So there seems to be no roadblocks for it to be safely integrated into Retropie (pi2). One good reason would be that IREM games look to be not supported at that time so this is a bit a bummer but a huge amount of games do work without any issues at all :)

Thanks for reading. If you need any input or tests to be done, just shoot.
Cheers,

TS

joolswills, 2015-08-30:
It will be added at some point yeh. thanks for the feedback.

nemo93, 2015-08-31:
thanks to YOU! I can't do a pull request for now but using the same setup as for the "old" core did work seamlessly. It just a matter of using the right repo as you'd have guessed :)

joolswills, 2015-09-03:
added the newer libretro fba port to experimental - it's called lr-fba-next - please try it out and let me know if it works well etc. 

Also how does the performance compare to the older fba - especially interested in rpi1 performance (I will do some testing of my own when I get time).

nemo93, 2015-09-05:
Thanks. It works like a dream. Turns out the Neo Geo Service Screen is also back in this version (without having to enable Unibios first).

I can only tell on rpi2 as I don't have rpi1 anymore. sorry about that. so on pi2 performance wise it looks very similar to the "old" one, I've seen no difference on Neo Geo and Capcom games (cps1/2).

nemo93, 2015-09-07:
The new experimental Mame based on (https://github.com/libretro/mame2003-libretro) adds support for those IREM games :)

closed 2015-09-07 by nemo93

thomasgoldstein, 2015-09-09:
Hi guys. Thanks a lot for this (and everything else), Jools! I just tried lr-fba-next (on Pi 2, don't have a Pi 1). From my tests: games don't run any faster or slower than with lr-fba. Slow games are still just as slow as before (e.g: Crude Buster), and full-speed games (SFA3) or nearly full-speed games (SF3) run just as well as before. This is still great for the NeoGeo, though (fixed sound issues, unibios 3.1, and no more black blinking when adding credits in some games, such as Money Puzzle Exchanger).

It seems to me that lr-fba-next just totally replaces the old lr-fba. The only "advantage" lr-fba has over lr-fba-next is that it doesn't expect the same romset for some games (e.g: for the sfiii.zip ROM, old lr-fba needs "sfiii_usa.29f400.u2", while the new one needs "sfiii_euro.29f400.u2"), so romsets that worked with lr-fba won't necessarily work with lr-fba-next.

joolswills, 2015-09-09:
ok . im convinced, thanks for the feedback. I will make the new core a default, and leave the old one as second, just in case.

barbudreadmon, 2015-09-13:
@thomasgoldstein this new fba core is in sync with official fba romset, unlike old core (never found an official fba romset with 100% compatibility, there was always some roms with a few missing files). that's definitely not an "advantage" for the old core and was one of the primary reason that got me started on this project.

I confirm it shouldn't be faster (the profile=performance hack is to get SH-2 emulator old behavior, new one is more accurate but will run poorly on rpi2), but at least it fixes some issues and add a few things over old core.
 
Perhaps retropie users will stop complaining on the new core github about old core issues... I talked a few times about this new core on your forums when i started the project, but never got a reply.

thomasgoldstein, 2015-09-15:
@barbudreadmon I agree that's it not actually an advantage, hence the quotes. It's just that some people might wrongly assume there might be bugs or regressions preventing their old ROMs from working, while they just don't have the right ones.

Anyway, I played around with lr-fba-next some more, and found many new games work (all Super Kaneko games, Alien Storm...), and some games that worked before but were buggy now work better (Altered Beast, Momoko 120%, and other games that had wrong colors). So thanks a lot for your work.

BillyTPilgrim, 2015-09-17:
Hi there,

I know this is a closed issue, but I just wanted to check - is there any progress on getting the NVRAM to save properly, for example to turn on blood in Metal Slug? I have tested it and it still seems to be broken, but I wanted to make sure there wasn't a setting that needed enabling or something.

alfredox123, 2016-05-17:
i tested it on an OC (turbo option) rp1 and all games that i test it runs between 40 and 50 fps, wich is unplayable games tested list:

- armored warrior
- vampire savior
- super street fighter 2 turbo
- street fighter alpha 3
- marvel vs capcom
- xmen vs street fighter
- marvel super heroes
- battle circuit (i think this one is playable between 50 and 55 fps)
- dungeon and dragons shadow aover mystara (this one was playable to i think)
- alien vs predator

barbudreadmon, 2016-05-17:
Yes, cps2 games (and probably a few others who use a highly clocked m68k) won't run at full speed on neither the new nor the old core on rpi1. That's a known fact for years. You can either buy a modern rpi, or try https://github.com/libretro/fba_cores_cps2 (i would be interested to know if cps2 games would run at full speed on rpi1 with this one). If you are a developper, you can also help https://github.com/libretro/libretro-fba , having an alternate 68k dynarec (something like cyclone, or perhaps c68k) would be nice.

alfredox123, 2016-05-17:
this version of pifba run those games at full speed on rp1 https://github.com/recalbox/pifba (it works on retropie os to), which is based on this version https://github.com/RetroPie/pifba that doesnt work with dpad, even though the change/update list says it does, thats why im using recalbox pifba which supports dpad

but theres a problem with second player that i put here https://github.com/recalbox/pifba/issues/2 (read the last comment) and it seems that it is abandoned? i dont know. the guy did respond, but didnt understand what he trying to say on first place and since then, he never respond again :(

barbudreadmon, 2016-05-17:
Yes, pifba is faster on 68k games, because it uses cyclone (a 68k dynarec for arm)

alfredox123, 2016-05-17:
is there a way to fix the problem im having with the second player?

barbudreadmon, 2016-05-18:
As i said, try https://github.com/libretro/fba_cores_cps2 (which use c68k and should be lighter/faster) or buy a new rpi, i can't help about pifba.

alfredox123, 2016-05-18:
can you tell me step by step how to install it? generally when i download a core theres a download file on releases tha ti put directly on retropie, but i dont know how to install it with this one :S

joolswills, 2016-05-18:
Please use the forum for discussion on this.

alfredox123, 2016-05-18:
i just tested the fba cps2 core and runs exactly the same as lr fba that comes with retropie

barbudreadmon, 2016-05-18:
Too bad, i thought c68k would improve speed. It means rpi1 is slower than a Wii. You should buy a new model, you won't get much help about pifba which has been a dead project for a long time, i think 99.99% of raspberry users switched to rpi2/3 and libretro-fba (which is a lot better if you have the hardware). Good luck if you don't !

alfredox123, 2016-05-18:
well thats a shame considering that pifba works really well, the only problem is with second player and is just the dpad that doesnt work well, all other buttons works great.....

joolswills, 2016-05-18:
Please use the forum for further discussion as I have already asked. This discussion is not related to this ticket which was closed last year. Please don't make me have to lock it.

joolswills, 2016-05-18:
(That was primarily for @alfredox123 btw - we use this issue tracker for bug reporting, not for general discussion. A closed ticket is not the right place to bring up issues with an emulator)

barbudreadmon, 2016-05-18:
No problem, anyway i'm done with my answer ;)