RetroPie/RetroPie-Setup #988: [RPI2] Get rid of unused emulators files and directories?

nemo93, 2015-08-30:
Hi there,

as i'm very demanding regarding what's installed on my systems I like to keep them as clean as possible :) So I got 2 issues:

1. there are still some packages not required (I guess cupsd is still there) that could be removed without any trouble (unless you want print something).
2. I'm not using the Atari/MSX/Coco/etc for now. And they are kind of "polluting" my Samba share and configurations. Is there any way to be able to remove them selectively? and make sure all files/directories for these emulators are safely removed? 

If I could do anything to assist please let me know.
cheers,

TS

joolswills, 2015-08-30:
Hi,

Support requests would be best posted at the forum - http://blog.petrockblock.com/forums/forum/retropie-project-forum/ where you are more likely to get help - we tend to keep this area for bugtracking etc.

But in answer to your questions

1. There is an option under supplementary / raspbian tools which removed some uneeded packages. You can manually remove more as you like

2. you can just remove the rom folders if you dont need them. You can also manually remove the installed emulators from /opt/retropie but there is little point really imho. It is also possible using retropie-setup and menu 5 to install individual components etc.

nemo93, 2015-08-31:
hi,

oh sure thing. I'll post at the forum then. apologize :( and thanks for answering anyway!

closed 2015-08-31 by nemo93