RetroPie/RetroPie-Setup #989: Lack of CEC on Kodi's Retropie

Tacioandrade, 2015-08-30:
Good evening, I was from the beginning of the week trying to put to work the CEC (control via remote
control) on Kodi's Retropie, however it would not in any way sought in various forums and no one
could answer me why the Periphericals item in System menu will not be enabled.

Now in the middle of the night in Brazil, I managed to figure out the solution, was missing
libcec. Then I would ask that the next version of Retropie menu, add the part of the Kodi
installation, the installation of the following libs: libcec1 and libcec2.

After installing both the CEC worked automatically on my Kodi.

I thank you again for the great work on the project that has greatly improved every day and I
hope this tip I'm giving help to make it even better.

Sincerely Tácio Andrade.

gizmo98, 2015-09-30:
Ok. Installed both packages and cdc is working. Thank you!

closed 2015-09-30 by gizmo98

HerbFargus, 2015-09-30:
This will work until kodi is updated to kodi 15 then we may have to look into libcec3

gizmo98, 2015-09-30:
Do you think we should already add libcec3 as well?

HerbFargus, 2015-09-30:
Idk its possible they may conflict. May also need cec-utils- 

taking my info from when I tested this build:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=109088

gizmo98, 2015-09-30:
All right. So if our raspi kodi package is updated to a version >= 15 it is certainly necessary to replace libcec1/2 with libcec3.

HerbFargus, 2015-09-30:
From what I understand yes. But Michael gorvens repo has been down for a few days now (at least the website has) so I'm hoping that means he's doing updates but we'll see. It also may be worth compiling ourselves pending on Michael gorvens repo having joypad support enabled in the build flags as it would be nice to have gamepad support in kodi.