RetroPie/RetroPie-Setup #991: Sam Coupe emulator

gonzo-the-great, 2015-09-02:
hi, would it be possible to add this emulator to the experimental menu, if it can be run on the Pi that is,

http://www.simcoupe.org/download/

thanks

gonzo-the-great, 2015-09-02:
yeah it boots, not tried anything else, but please could you add it to the experimental menu

thanks

gonzo-the-great, 2015-09-03:
ok just loaded up SAM cards.sad, loads up nice and quick, but the mouse is very jumpy, which means its very hard to control, sadly, but on i'll test & play

gonzo-the-great, 2015-09-03:
oh, btw, heres the SimCoupe.cfg, suggestions how to make the mouse less choppy/jumpy?

http://pastebin.com/tfA70z2a

i have it running on one of my rpi B+ boards overclocked to moderate

gonzo-the-great, 2015-09-11:
alls cool on a rPi 2!!

gonzo-the-great, 2015-09-15:
has anyone else been able to try this out yet?

joolswills, 2015-10-10:
Reference to source for myself - https://github.com/simonowen/simcoupe/

HerbFargus, 2016-03-05:
Preliminary Module:

http://pastebin.com/2echyVVC

joolswills, 2016-04-09:
Thanks - added in 11e797da04ed464b25a8f6115a6110c8d6d74277

closed 2016-04-09 by joolswills