RetroPie/RetroPie-Setup #993: Samba customs shares are lost when updating RetroPie

modem, 2015-09-03:
Hello,

I've been loosing my samba custom shares everytime I update RetroPie on Raspbian.
I updated yesterday to from v3.0 RC1 to v3.0 and my smb.conf file was changed from: 

```
[global]
    workgroup = WORKGROUP
    #usershare allow guests = yes
    #security=share
    security=user
    follow symlinks = yes
    wide links = no
    unix extensions = no
    lock directory = /var/cache/samba
    socket options = TCP_NODELAY IPTOS_LOWDELAY SO_RCVBUF=65536 SO_SNDBUF=65536
    write cache size = 2097152
    getwd cache = yes
    min receivefile size = 16384
[pi]
    browsable = yes
    read only = no
    #guest ok = yes
    valid users = pi
    path = /home/pi
    #force user = pi (no longer needed)
[devices]
    browsable = yes
    read only = no
    #guest ok = yes
    valid users = pi
    path = /media
    force user = root

[roms]
comment = roms
path = /root/RetroPie/roms
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root
[bios]
comment = bios
path = /root/RetroPie/BIOS
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root
[configs]
comment = configs
path = /opt/retropie/configs
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root

[HDD1]
comment = My External HDD
path = /mnt/HDD1
read only = No
create mask = 0777
directory mask = 0777
guest only = Yes
guest ok = Yes

[HDD2]
comment = External HDD2
path = /mnt/HDD2
read only = No
create mask = 0777
directory mask = 0777
guest only = Yes
guest ok = Yes
```

to 

```
[global]
    workgroup = WORKGROUP
    #usershare allow guests = yes
    #security=share
    security=user
    follow symlinks = yes
    wide links = no
    unix extensions = no
    lock directory = /var/cache/samba
    socket options = TCP_NODELAY IPTOS_LOWDELAY SO_RCVBUF=65536 SO_SNDBUF=65536
    write cache size = 2097152
    getwd cache = yes
    min receivefile size = 16384
[pi]
    browsable = yes
    read only = no
    #guest ok = yes
    valid users = pi
    path = /home/pi
    #force user = pi (no longer needed)
[devices]
    browsable = yes
    read only = no
    #guest ok = yes
    valid users = pi
    path = /media
    force user = root

[roms]
comment = roms
path = /root/RetroPie/roms
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root
[bios]
comment = bios
path = /root/RetroPie/BIOS
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root
[configs]
comment = configs
path = /opt/retropie/configs
writeable = yes
guest ok = yes
create mask = 0644
directory mask = 0755
force user = root

```

It seems there is a bug in the update process. Can you please check it?

joolswills, 2015-09-03:
The code assumes some old capitalized shares were created by itself and it removes them. It was never really designed to work too well with custom user edited samba configs. If you change your shares to lowercase it won't remove them. I may improve the old share removal code if I get a chance - but note that upgrading retropie 3.0rc1 -> 3.0 does require manual work and some config merging etc too.

joolswills, 2015-09-03:
I may just remove the old samba share removal code actually, as no-one should be updating from such an old version as when that was needed.

joolswills, 2015-09-03:
ok - that's done now - it shouldn't touch those shares. It can still overwrite shares that collide with its names though - but it should be safer with your custom config.

modem, 2015-09-03:
Thanks for the quick fix.

joolswills, 2015-09-03:
welcome.

closed 2015-09-03 by joolswills