RetroPie/RetroPie-Setup #995: retropiemenu and ES-Scraper

xBurningHero, 2015-09-03:
retropiemenu is now missing from my EmulationStation menu. 

Yesterday I was configuring RetroPie to find roms on my USB Jump Drive. I had no internet access and a really terrible battery powered keyboard, so in order to set it up more quickly I performed a mass `sed` on `/etc/emulationstation/es_systems.cfg` replacing `~/RetroPie/` with `/home/usb0/`
Which of course broke the entry for 'retropie'
I reverted the filepath in es_systems.cfg for retropie so it now points to ~/RetroPie/retropiemenu however no matter what I do I cannot get the retropie menu to reappear in ES. All of the rp scripts still exist in `~/RetroPie/retropiemenu/`
Is there any script that can be run to check for missing components or some way to fix this without starting over?

Additionally, I am trying to run ES-Scraper at nights while I sleep. It runs really slowly and so it only goes through a few dozen before the device seems to go to sleep. I can watch it do 50 or so, and then go to bed, but once I wake up it's only at 55 or so. Something keeps putting it to sleep. Is there a way to run ES-Scaper in command line or force it to not sleep or disconnect if the system is idle?

(Sorry for asking here rather than in the FAQ but I couldn't find any references to either on the forum)

joolswills, 2015-09-03:
you can do

```
sudo ./retropie_packages.sh retropiemenu configure
```

to re-set up the ES entry for the menu. 

in regards to roms, instead of messing with ES configuration, it would be better to bind mount the usb drive over the original rom location or symlink it or something.

instead of ES scraper - you can update retropie-setup and try out the sseph's scraper (in experimental). better to run it when ES isn't loaded though as there might be conflicts if running it when ES is running (ES may overwrite configs on exit)

Note: this issue tracker is for bugs, not support - please use the retropie forum for general help.

xBurningHero, 2015-09-03:
Understood. Thank you for your quick response and assistance.

closed 2015-09-03 by xBurningHero