RetroPie/RetroPie-Setup #996: px68k - SHARP X68000 Emulator

gonzo-the-great, 2015-09-04:
i've just found this, px68k its a SHARP X68000 Emulator, i was thinking of giving it a go, but the post is in Japanese, I've tried an online translation thing, but they make no sense!

heres the post...
http://hissorii.blog45.fc2.com/blog-entry-238.html

heres the github page, in Japanese too...
https://github.com/hissorii/px68k

joolswills, 2015-10-07:
I have it runnning - performance is not good - getting only a few FPS.

joolswills, 2015-10-07:
Now available under the experimental menu.

gonzo-the-great, 2015-10-07:
wow! you guys are great! thanks!
now i feel bad for wasting all your time, if its only a few fps....

sorry

will still test!

joolswills, 2015-10-07:
I guess it would need some arm optimisations to be usable - like an arm optimised 68000 emulation core.

closed 2015-10-08 by joolswills